Bodmer/TFT_eSPI #1320: Touch does not work on ILI9341 with ESP32

LearnCoding404, 2021-09-01:
Hello TechWorld,

I am new here in TFT_Espi programming and my problem is when I load the code on the ESP32 i see that the touch is oke. 
[image]

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
`#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
//====================================================================
void setup(void) {
  Serial.begin(115200);
  Serial.println("\n\nStarting...");
  tft.init();
}
//====================================================================
void loop() {
  uint16_t x, y;
  tft.getTouchRaw(&x, &y); 
  Serial.printf("x: %i     ", x);
  Serial.printf("y: %i     ", y);
  Serial.printf("z: %i \n", tft.getTouchRawZ());
  delay(250);
}`
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

But if i use the samples from library ( Keypad_480x320 and On_Off_Button)  code the Touch doesnt do anything!

I need your help please :(

daverathbone, 2021-09-02:
I had lots of problems with Touch and TFT-eSPI pin conflicts. When using 8 bit parallel displays. 

I now use ESP32_TouchScreen libary (Google it) and that works. 
But is very Display dependant as diffrent displays that look the same on Amazon (UK) have diffrent Touch screen Pin outs.  
I think that China markets are mainly for Mega 2560 as host for these low cost display types.
Which can use touch I/O pins that would conflict with the ESP32 ADC pin outs.
Problem is this class of displays are not back in large production yet,  so volume product design be aware!
I would have used SPI but supply is almost zero.

The resistive touch types  are very poor as its coating is not even. 
So that you can get crazy values for x,y at diffrent points accross the display.
With very poor readings near edges.

However if you make "Human" big press buttons for your appliction it works. (Forget QWERTY!  key buttons)
I'm using ESP32 uno clone made by Az-Delivery also from Amazon (UK) as my host for the 480x320 LCD touch screen
I had to buy a few from diffrent Amazon supliers  to get Touch pin outs I needed.
(plugs direct into board see Bodmer's extra wire fixes)  Set up as:

Display:
#define ILI9488_DRIVER  (Make sure your chip on glass is this!!!! China suppliers don't care on adverts)
#define TFT_PARALLEL_8_BIT
#define TFT_CS   33  
#define TFT_DC   15  
#define TFT_RST  32 
#define TFT_WR   4  
#define TFT_RD    2  
#define TFT_D0   12  
#define TFT_D1   13  
#define TFT_D2   26  
#define TFT_D3   25  
#define TFT_D4   17  
#define TFT_D5   16  
#define TFT_D6   27
#define TFT_D7   14
 
Resistive Touch :
const int XP = 27, XM = 15, YP = 4, YM = 14, YP2=33;        

Most important in code  after reading touch (Get touch) is to return I/O back to TFT display
pinMode(YP, OUTPUT);      
pinMode(XM, OUTPUT);
digitalWrite(YP, HIGH);   
digitalWrite(XM, HIGH);

All taken many hours and money to get a number of device's working for my Client
There are still some bugs arround  tft.setTextDatum(MC_DATUM);
Which make other screen text jump?  
Poor use of  text handlers that leave last text behind .....
but I'm thankfull to Bodmer for his work.

dannybackx, 2021-09-09:
Does work for me, see https://sourceforge.net/p/espalarm/svncode/HEAD/tree/main/

Bodmer, 2021-09-13:
I suspect this is down to touch calibration. If you do not use the calibration value for the screen orientation you final sketch uses then the touch will be recorded in the wrong place. Add some debug print lines to your sketch to report touch coordinates.

copyCATandpasteDOG, 2021-10-02:
Did you erase the SPIFFS before a new caljbration?
That's what i always do.

daverathbone, 2021-10-14:
It would be handy if the users that reply here , could do so with some more detail about what was achived.

Bodmer, 2021-11-28:
@LearnCoding404 
I asume this has been resolved. I have not been able to identify any issue with TFT_eSPI.

closed 2021-11-28 by Bodmer