Bodmer/TFT_eSPI #1321: 2.3.70 incompatible to TTGO T8 v1.7 ESP32?

LiveRock, 2021-09-04:
Hi

I have the TTGO T8 v1.7 ESP32 board.
I tried to update TFT_eSPI to the latest but cannot as it is dimmed out in VSCode with PlatformIO.

[image]

Why is it incompatible?

Bodmer, 2021-09-13:
I do not use Platformio but other users have had success. Maybe this is down to the environment settings?

maciekr1234, 2021-09-15:
you probably specified exact version in library dependencies try either removing the version specifier or set it to @^2.3.70.

you can read more about semanting versioning [here](https://devhints.io/semver).

Bodmer, 2021-09-18:
@maciekr1234 
Seems reasonable, will wait for feedback.

LiveRock, 2021-09-20:
Thanks. It works. I just need to manually update platformio.ini

closed 2021-09-20 by LiveRock