Bodmer/TFT_eSPI #1322: Drivers for R61505

m-menon, 2021-09-05:
hi,

Any chance of including this driver ([R61505](https://datasheetspdf.com/datasheet/R61505.html)) in the future? I came across a bunch of boards (TFT IPS type display) that use this driver IC. Wanting to try out the IPS type display because of it's better viewing angle. 

Thanks for the amazing job with this library.

Bodmer, 2021-09-13:
Provide a link to the display vendors product page. SPI or 8 bit parallel?

m-menon, 2021-09-13:
SPI / 4 wire. On a deeper look it doesn't seem like a very popular chip but I got a couple of links. If you think it isn't worth the effort then pls feel free to close the issue as one can easily find alternative boards with more popular, supported chips like the ILI9341, ILI9488 etc.

https://www.aliexpress.com/item/32244019358.html ( the [thing](https://www.aliexpress.com/item/32244019358.html) is cheap though )
https://www.ramtex.dk/display-controller-driver/rgb/r61505.htm
[Renesas Datasheet](https://datasheetspdf.com/pdf-file/928573/Renesas/R61505/1)

Bodmer, 2021-09-14:
Thanks for the links. I would need to buy one to be sure to get the driver working as sometimes the driver does not work for various bug/functional misunderstanding/timing reasons. If I don't test it I get user issues raised which takes more effort in the long term!  Glad you appreciate the effort it can take!

The screen is quite cheap and thus attractive for users.  It may be that this driver becomes more popular as I notice the supplies of cheaper ( < $10) displays is dwindling as the market demands move on. As you say there are lots of alternatives.

closed 2021-09-14 by Bodmer

m-menon, 2021-09-14:
You've done an amazing job with the library. If you don't mind I can pay for a couple of these boards (PayPal is fine?). I have no immediate need because I ordered a few other boards but as you said, this might help a lot of others. So please do let me know if I can pay for the boards. I am not sure if AliExpress sells vouchers or something in which case it would make it easier, or alternatively PayPal works.

Bodmer, 2021-09-14:
Thanks for your offer, I have other priority projects at the moment and thus have to decline as it is a matter of available time rather than cost. I will add to my TODO list but that is very long!

m-menon, 2021-09-14:
No worries. These things are seriously time consuming.