Bodmer/TFT_eSPI #1326: TFT_eSPI -  ESP32-S2 - ST7789

sbonaime, 2021-09-07:
Hi,

As Arduino ESP32 Core 2.0.0 is out, I would like to use TFT_eSPI for my ESP32-2 Saola and my adafruit 240x320 ST7789.
With the same hardware, I can use Adafruit_ST7789 driver using

```Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);```

But with TFT_eSPI nothing is display.

Here is the result of Read_user_setup.ino

```
10:35:09.139 -> [code]
10:35:09.139 -> TFT_eSPI ver = 2.3.70
10:35:09.139 -> Processor    = ESP32
10:35:09.139 -> Frequency    = 240MHz
10:35:09.176 -> Transactions = Yes
10:35:09.176 -> Interface    = SPI
10:35:09.176 -> Display driver = 7789
10:35:09.176 -> Display width  = 240
10:35:09.176 -> Display height = 320
10:35:09.176 -> 
10:35:09.176 -> MOSI    = GPIO 35
10:35:09.176 -> MISO    = GPIO 36
10:35:09.176 -> SCK     = GPIO 37
10:35:09.176 -> TFT_CS   = GPIO 34
10:35:09.176 -> TFT_DC   = GPIO 33
10:35:09.176 -> 
10:35:09.176 -> Font GLCD   loaded
10:35:09.176 -> Font 2      loaded
10:35:09.176 -> Font 4      loaded
10:35:09.176 -> Font 6      loaded
10:35:09.176 -> Font 7      loaded
10:35:09.176 -> Font 8      loaded
10:35:09.176 -> Smooth font enabled
10:35:09.176 -> 
10:35:09.176 -> Display SPI frequency = 36.00
10:35:09.176 -> [/code]
```

My User_Setup_ESP32-S2.h is
````
#define ST7789_DRIVER // Full configuration option, define additional parameters below for this display
#define TFT_RGB_ORDER TFT_BGR // Colour order Blue-Green-Red
#define TFT_WIDTH 240 // ST7789 240 x 240 and 240 x 320
#define TFT_HEIGHT 320 // ST7789 240 x 320
#define TFT_MOSI 35
#define TFT_MISO 36
#define TFT_SCLK 37
#define TFT_CS 34 // Chip select control pin
#define TFT_DC 33 // Data Command control pin
#define TFT_RST -1 // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define LOAD_GLCD  // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2 // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4 // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6 // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7 // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8 // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
#define SMOOTH_FONT
#define SPI_FREQUENCY 36000000
#define USE_HSPI_PORT
````

With Test_readWrite I assume, I can write to the screen but not read. So I am not very sure, I wrote !
````
10:44:27.478 ->  Pixel value written = 80
10:44:27.478 ->  Pixel value read    = 0
10:44:27.478 ->  ERROR                 ^^^^
10:44:27.973 ->  Pixel value written = 100
10:44:27.973 ->  Pixel value read    = 0
10:44:27.973 ->  ERROR                 ^^^^
````
I try to change frequency without results. Pins definitions seem ok for me as they are the same I use with Adafruit_ST7789.
I also try the ST7789_2_DRIVER. I don't have any compilation error
I use ```#define USE_HSPI_PORT```  as  proposed in #807. Maybe there is something more

Do you have any ideas ? 

Thank you very much

sbonaime, 2021-09-07:
With ```#define ILI9486_DRIVER``` I can see something on the screen. I think, my problem is related to the driver. I will try other ones
[image]

sbonaime, 2021-09-07:
With #define ILI9486_DRIVER and SPI_FREQUENCY 32000000,  the TFT_Starfield is working at 35fps for 3/4 seconds with a black background, then it switches to white background with black start for also 3/4 secs. After that the screen turn to black. There is definitely a driver problem. But I think I am not too far to succeed !

JHershey69, 2021-09-10:
Ironically, the ST7789 was what I was working on a problem with a blank screen for a user and he sent me this message:
"I found it was a pin problem. Apparently, they moved MOSI TO 19 from 23."
Don't know if that will help.

sbonaime, 2021-09-10:
@JHershey69 I changed my MOSI, MISO and others pins to what I need. I am sure it is not the problem because with same wiring and hardware, I can use  Adafruit_ST7789 driver

Bodmer, 2021-09-12:
Run Read_User_Setup again and check output because your reported values quoted above do not tally...
```
10:35:09.176 -> Display SPI frequency = 36.00
```
and
```
#define SPI_FREQUENCY 8000000L
```

Bodmer, 2021-09-12:
```
TFT_Starfield is working at 35fps for 3/4 seconds with a black background, then it switches to white background with black start for also 3/4 secs. 
```
This suggests an inadequate power supply. Bear in mind TFT_eSPI can use high clock rates and this increases power consumption.

Bodmer, 2021-09-13:
I read your report again and see you have tried most options available. I can only conclude that this is a particular issue with the newer ESP32-S2 whic probably requires some changes to the library. I will have to get an ESP32-S2 board and  update the library.

Thinkpad4by3, 2021-09-14:
I have an ESP32-S2 with a HXD8357D with a nearly identical problem. I can't get the display to work(white screen) unless I use the 18-bit define flag, which it works with limited success and visual artifacting like shown above. The Adafruit library works just fine for the LCD. The pixel error count is exactly as shown above.

Bodmer, 2021-09-15:
The pixel read test will always fail on this display because Adafruit did not wire MISO to the TFT, it is only wired to the SD card slot. The Adafruit screen also uses a level translator buffer that may require a lower SPI clock rate to be used, perhaps that buffer introduces a timing skew so commands are getting corrupted.  Try 8 MHz and see what it can be raised to before problems occur.

sbonaime, 2021-09-22:
I tried with ``` #define SPI_FREQUENCY   5000000``` and I have the same problem. 

On thing to notice is that the ST7789 screen is not directly driven through its pins. There is  a "pixel-addressable frame buffer" in between. 

While doing several tests with ILI9486_DRIVER, I  sometimes see ghost of previous examples
[image]

zenmetsu, 2021-10-02:
Just tossing this out there, I recently acquired a display with an ST7789.  My prior display used an ST7735 and worked well.  This display does not show any image with either the ST7789 or the ST7789_2 drivers.  If I use the ST7735 driver, it kinda works, but i can't get the full screen to work, only a 240x240 subset of the pixels i think.  If I use one of the other drivers, like ILI9486 or ST7796, i can at least get an image on all 320x240 pixels, but it's mirrored and has a few other issues from time to time.

I can help with debugging this code as well since i have an ST7789 on hand to test with.  Pretty sure it is a driver bug at the moment and i've been going through the code and the datasheet for the past few hours to try sorting it all out.

zenmetsu, 2021-10-02:
For the time being, I can confirm that the following will work...

If you back up the driver files, and then overwrite the ST7796 files with the `ST7789_Defines.h`, `ST7789_Rotation.h`, and `ST7789_Init.h` versions, and set your User_Setup.h to use ST7796, it seems to work just fine.

zenmetsu, 2021-10-03:
I believe that i found the problem.

I used the ST7789 driver and changed only the `TFT_eSPI.h` file.  There is a section:
`// Some ST7789 boards do not work with Mode 0
#if defined(ST7789_DRIVER) || defined(ST7789_2_DRIVER)
  //#define TFT_SPI_MODE SPI_MODE3
  #define TFT_SPI_MODE SPI_MODE0
#else
  #define TFT_SPI_MODE SPI_MODE0
#endif`

By forcing Mode0, my board started working.

Now I am facing different problem, probably due to MCU choice, where SPI cannot be set over 29MHz, datasheet show 29MHz as maximum SPI clock for "slave" mode, so perhaps this library is instantiating SPI incorrectly on this board.

[image]

nikthefix, 2021-10-05:
For a quick and dirty fix for S2 compatibility do this:

Go to TFT_eSPI_ESP32.c in TFT_eSPI lib / processors folder.
Change line 12 HSPI to FSPI
Go to  User_Setup.h
Uncomment #define USE_HSPI_PORT

Go TFT_eFEX.h and change #ifdef ESP32 to ESP32S2 on line 17, 22, 54, 116
Go TFT_eFEX.cpp and change #ifdef ESP32 to ESP32S2 on line 676, 1389

Done!

sbonaime, 2021-10-06:
Actually with "Meters" example, the screen is right-left mirrored and I have some greenish colors.

[image]

nikthefix, 2021-10-06:
Did you try TFT_inversion at user_setup.h lines 107, 108?

Egon466, 2021-10-07:
Hello, today I have used the same display Adafruit HX5387D on three different processor, DOIT ESP32 DEVKIT V1 as a 8 bit parallel connection, RPI RP2040 SPI and Nucleo-F466RE SPI. The ESP32 displayed TFT_Meter correctly. The RP2040 displayed the TFT_Meter demo Mirrored, as show by sbonaime above, and the text was distorted. The Nucleo-F466RE displayed TFT_Meter exactly like sbonaime results.

I have used the same HX5387D in both parallel and SPI mode on various other systems with positive expected results.

The Nucleo-F466RE is the current setup and would really like some help with getting it going. The only file that I have modified in either library or example is the User_Setup.h which I have attached.

Thank you.
[User_Setup.h.zip](https://github.com/Bodmer/TFT_eSPI/files/7301617/User_Setup.h.zip)
.

Egon466, 2021-10-09:
OK, on the RD2040 I tried 10MHz with now change, Still Mirrored. As I don't know if setting SPI_FREQUENCY has any affect because of the Definition RPI_DISPLAY_TYPE I commented out RPI_DISPLAY_TYPE and it worked with a SPI_FREQUENCY of 10MHz. 

I then set SPI_FREQUENCY back to 20MHz and it still worked. 
Then un-commented RPI_DISPLAY_TYPE and Mirrored again!
Commented out RPI_DISPLAY_TYPE and correct display.
Upped SPI_FREQUENCY to 40MHz and still correct display.

Dose SPI_FREQUENCY change anything on a RPI RD2040? Not that it really concerns me as it works with the current setting and that is the main thing.

I will play around with the SPI settings on the NUCLEO tomorrow.

Egon466, 2021-10-10:
I had three problems with two processors on the Adafruit HX8357D.

Problem 1. Raspberry Pi Pico RD2040 with HX8357D, Dynamic Text either not being displayed or distorted. This was fixed by selecting a different IDE core Board processor definition. Originally I selected the "Arduino Mbed OS on RD2040" which is all I had installed on the IDE at the time. When I installed and changed to "Raspberry Pi RD2040 Boards (1.9.5)" Board definition that solved this Problem.

Problem 2. Raspberry Pi Pico RD2040 with HX8357D, Mirrored display in all rotations. This was fixed by commenting out the definition RPI_DISPLAY_TYPE in User_Setup.h. Also successfully ran Raspberry Pi Pico SPIO to 40MHz.

Problem 3. Nucleo-F446re with HX8357D,  Mirrored display in all rotations. This was fixed by inserting tft.begin(); after tft.init(); in setup() of example "TFT_Meters.ino".

These are the only problems I had. 
Regards,
Egon

Bodmer, 2021-11-18:
The library now supports the ESP32 S2, this has been tested with the Arduino ESP32 board package 2.0.1.

A new setup file example has been added:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70_ESP32_S2_ILI9341.h

closed 2021-11-18 by Bodmer