Bodmer/TFT_eSPI #1327: R61529 parallel with capacitive touch - bad sensitivity

dexter323i, 2021-09-07:
Hi! Thank you for your library, it is so much useful for me! It works perfectly with R61529 in parallel mode with the configs I include below. The display comes with a capacitive touch screen, it's driver is FT6206. It is communicating by I2C / Wire.h. I have a very strange issue:
When I use only the touch screen, it is quick, accurate, responsive. But as soon as I call tft.begin() (before or after touch begin, does not matter), the touch sensitivity goes bad. It does not register the touch many times , I have to touch it 4-5-6 times to recognize one touch. Then it is okay. Then bad again, or reads touch from a totally bad location, or negative value! If I comment out the tft.begin() it works perfectly again. I could find exactly where is the code what ruins it: In TFT_eSPI.cpp in function TFT_eSPI::init(uint8_t tc).
This part runs okay:
```
delay(150); // Wait for reset to complete
begin_tft_write();
tc = tc; // Supress warning
```
But the following
`#include "TFT_Drivers/R61581_Init.h"`
ruins it.

In this file, the first line is:
`writecommand(TFT_SLPOUT);`
If this runs, my issue occurs.

I could copy together all the commands from the defines, that runs in this sequence, and what breaks the touch sensitivity. I put these after the tc = tc; line:
```
GPIO.out_w1tc = (1 << 4);
GPIO.out_w1tc = (1 << 14);
GPIO.out_w1tc = (((1 << 18) | (1 << 19) | (1 << 21) | (1 << 22) | (1 << 23) | (1 << 25) | (1 << 26) | (1 << 27)) | (1 << 13)); 
GPIO.out_w1ts = 0x840000; // originally:  xset_mask[0x11];
GPIO.out_w1ts = (1 << 13);
```
If I comment out the last line, or the line before the last, or both, then the touch is okay. But if I leave them both to run, the touch goes bad.

ESP32 WROVER 16MB

Touch screen connection:
1	VDD		- 3.3V
2	SDA		- IO12
3	SCL		- IO2
4	INT		- 10K pullup to 3.3V
5	RST		- 10K pullup to 3.3V
6	GND	- GND

User_Setup.h:
```
#define R61581_DRIVER
#define ESP32_PARALLEL

// Pins
#define TFT_CS  4   // Chip select control pin 
#define TFT_DC  14  // Data Command control pin - must use a pin in the range 0-31 
#define TFT_RST 32  // Reset pin
#define TFT_WR  13  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD  5  

#define TFT_D0   18  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   19  // so a single register write sets/clears all bits
#define TFT_D2   21
#define TFT_D3   22
#define TFT_D4   23
#define TFT_D5   25  
#define TFT_D6   26
#define TFT_D7   27  
```

Could you please, PLEASE help me what is wrong? Why the TFT conflicts with the touch? And how could fix it?

Bodmer, 2021-09-12:
The TFT_eSPI library has no built in support for the R61529 so that is likely to be the main reason for the problems. Are you trying to patch the library for that display?

dexter323i, 2021-09-12:
@Bodmer  not trying, but patched, and using continuously. :)
I researched furthermore this FT6206 touch issue, and I suspect there is a reason that the more commonly supported FT6236 touch layer costs $6.6 while this FT6206 costs only $4.4... 

I added two 4.7K pull up resistors to touch SDA and SCL data lines, what broke ESP32 FW upload via USB. (Platformio / VS code could not start communication with the ESP32) I think because IO2 and IO12 pins are strapping pins. So the next I tried is swapping some pins: Connected touch SDA and SCL to the default ESP32 I2C pins: IO21 and IO22. And moved TFT D2 and D3 lines to IO2 and IO12. After this, the touch started to work OK (not perfect, but useable), and the TFT also worked still.

The end result is: everything I need is working, so I guess this issue can be closed, because nothing wrong with TFT_eSPI.

p.s.: Still waiting for the official R61529 support. :)

closed 2021-09-12 by dexter323i

Bodmer, 2021-09-13:
Maybe the problem is in the FT6206 library you are using. I have used the FT5206  and created a [library](https://github.com/Bodmer/Touch_FT5x06) for it as I found others to be unreliable.

Glad you have it mostly working well enough.