Bodmer/TFT_eSPI #1328: M5Stack - TFT LED voltage too high!

Sarah-C, 2021-09-08:
Hi everyone!

I noticed when using this library the M5Stack #include setup is running the illumination LED at a much higher level than that advised by M5Stack engineers. 

I'd asked them previously about running the device at 3V3 rather than the 2V8 the M5Stack drivers use, for sunny conditions and got this reply back:

```
Me:
Is 2.8 volts for the backlight a "Comfortable viewing brightness", and isn't the maximum safe voltage it can be run at?
And is the 3.3 volts allowed by SetLcdVoltage not recommended, due to the LED getting hot/shortening the white LED life of the screen?
Simply put - is it safe to run the backlight at 3.3 volts all the time, for use in sunny conditions?

------------------------------------

M5Stack (support) <support@m5stack.com>
Hi Sarah,
No problem for short-time running in 3.3V. Let's say 1 or 2 hrs.
Not recommend for long-time running. Due to heat and life time concerns.

```

Therefore the M5Stack setup - the LED voltage should be reduced to their standard 2V8 to prevent over-heating.

m-menon, 2021-09-08:
I am not familiar with M5Stack but I guess this issue [here](https://github.com/Bodmer/TFT_eSPI/issues/226) should give you some pointers or ideas.

Sarah-C, 2021-09-09:
> I am not familiar with M5Stack but I guess this issue [here](https://github.com/Bodmer/TFT_eSPI/issues/226) should give you some pointers or ideas.

I've posted some code for people to use in their M5Stack sketches for "Core" and "Core 2" units.

Bodmer, 2021-09-12:
The TFT_eSPI library has NO built in support for back light LED brightness control. I suspect that is part of the M5Stack library and there should be reported in that repository.

Ideally M5Stack should design their products so that they cannot be damaged by the user software setting.

closed 2021-09-12 by Bodmer

Sarah-C, 2021-09-14:
@Bodmer   The initialisation code of TFT_eSPI initialises the backlight to max brightness:
https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9341_Init.h

That is not a user software setting, but is the driver code for the screen that replaces the libraries M5Stack uses. That's why this issue is relevant. It even overrides the M5Stack library if included!

Therefore using this library with the M5Stack option within the #defines will damage it. 
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h 
//#include <User_Setups/Setup12_M5Stack.h>           // Setup file for the ESP32 based M5Stack

You SHOULD NOT be suggesting this library is compatible with the M5Stack then.

Bodmer, 2021-09-14:
I have looked at this further and see the situation.

Essentially TFT_eSPI has evolved and so has the product range in the M5Stack brand so the original M5Stack module has been renamed "M5Stack Basic Core" so the new one can be given a new name "M5Stack Core 2". This means the TFT_eSPI setup file is for the original "M5Stack Basic Core" released in 2017.

I also see the M5Stck Core 2 product has a power managment chip for the backlight, connected via I2C and from your description that appears to default to maximum (potentially damaging) backlight brightness if it is not configured by a specialist APX192 library with specific configurations. This is not a problem TFT_eSPI should be adapted to resolve.

You need to look at the M5Stck Core 2 product data sheets as the TFT SPI interface pins for the M5Stck Core 2 are also different to the M5Stack Basic core. It will then be easy to generate a compatible setup file for it. When you have one then post it and I can added it to the library, please add comments and web links for other libraries that will be needed for the APX192, touch controller etc.

To resolve the ambiguity in the mean time I will rename "Setup12_M5Stack.h" to "Setup12_M5Stack_Basic_Core.h".

Sarah-C, 2021-09-14:
>To resolve the ambiguity in the mean time I will rename "Setup12_M5Stack.h" to "Setup12_M5Stack_Basic_Core.h".

That's a great idea. Thanks for watching out for users of the later devices. =)

> You need to look at the M5Stck Core 2 product data sheets as the TFT SPI interface pins for the M5Stck Core 2 are also different to the M5Stack Basic core. It will then be easy to generate a [...]

I will! It's quite the responsibility so I'll do my best.

Bodmer, 2021-09-14:
@Sarah-C 
> 
> I've posted some code for people to use in their M5Stack sketches for "Core" and "Core 2" units.

Where is this posted? Please provide a link.

Sarah-C, 2021-09-14:
https://github.com/Sarah-C/M5Stack_Core2_ScreenBrightness
It's not a library - just a simple function a user at any level can paste into their code.
It uses M5.Axp.SetDCVoltage and M5.Axp.SetDCDC3 - those functions will need to be extracted for a completely "stand alone" function.

I've lost my Core code, I'll have a look.

Bodmer, 2022-05-15:
There is an interesting discussion here from @belm0 about the M5Stack Core 2:
https://github.com/Bodmer/TFT_eSPI/pull/1815
This indicates that the backlight over drive problem is related to incorrect configuration of the AXP192 by the user sketch. The TFT_eSPI library has no functions to control the AXP192 power functions, so the library does not cause over drive of the backlight.

TFT_eSPI can therefore be used safely with the Core 2 providing the user adds the AXP192 code to their sketch and correctly configures the device. Refer to the M5Stack Core 2 documentation for details.

dizcza, 2023-03-11:
I've come to this issue from the discussion https://community.m5stack.com/topic/4278/stand-alone-tft_espi-library-on-core2/4 which is the second result I googled.

I believe many other users would like to have support for M5Core2 to be here. This would allow us not to use their M5Core2 display library but instead use TFT_eSPI lib which is superior and has broad board coverage.

Tried the suggestions from the link above and it worked fine. And before that, I spent 45 minutes trying to understand why `#include <User_Setups/Setup12_M5Stack_Basic_Core.h>` does not work with *any* examples in the lib, as I erroneously thought that this header file is both for M5Stack and M5Core2 (as it is the case for some other libs).

hpsaturn, 2023-10-08:
I found some improvements around that. I don't understand why this is not included in the UserSetup config, and some PRs was declined. Never mind, I found the next config for the **M5Core2 AWS** version:

```python
 -DUSER_SETUP_LOADED
  -DILI9342_DRIVER=1
  -DTFT_INVERSION_ON=1
  -DTFT_MISO=38
  -DTFT_MOSI=23
  -DTFT_SCLK=18
  -DTFT_CS=5
  -DTFT_DC=15
  -DTFT_RST=-1
  -DTOUCH_CS=-1
  -DLOAD_GLCD=1
  -DLOAD_FONT2=1
  -DSPI_FREQUENCY=40000000
  -DSPI_READ_FREQUENCY=8000000
``` 
Also I'm using other features without the M5Core2 library. Only some Touch headers and the Axp192 driver. More details in [this PR](https://github.com/atomic14/esp32-tv/pull/10) of other project that using that.  And works fine:

https://github.com/Bodmer/TFT_eSPI/assets/423856/bccf6b07-f434-44fa-ab18-83bedf54bf0c