Bodmer/TFT_eSPI #1335: ILI9341 Not working black screen

adolfonovo, 2021-09-10:
Hello
I have a 2.4 AzDelivery TFT display (8bit parallel) and I can´t get it work with the TFT_espi library.
The manufacter says that is controlled by ILI9341 driver, and I test it with an Uno board and MCUfriend detects as a 0x9341, so I assume it is ok. Also all the examples with the UNo and mcufriend are working

I connect the ESP32 and define the user setup as follows:

```
// See SetupX_Template.h for all options available

#define TFT_PARALLEL_8_BIT
#define SPI_FREQUENCY  40000000
//#define SPI_TOUCH_FREQUENCY  2500000
//#define ESP32_PARALLEL

//#define TFT_SDA_READ
//#define NUCLEO_64_TFT

//#define R61581_DRIVER //3.5 TFT
//#define ILI9481_DRIVER
#define ILI9341_DRIVER
//#define ILI9341_2_DRIVER
//#define ILI9488_DRIVER
//#define ILI9486_DRIVER
//#define ILI9163_DRIVER
//#define ST7735_DRIVER
//#define S6D02A1_DRIVER
//#define HX8357D_DRIVER
//#define ST7789_DRIVER

// ESP32 pins used for the parallel interface TFT
#define TFT_CS   33//azul  // Chip select control pin
#define TFT_DC   18//verde  // Data Command control pin(LCD_RS) - must use a pin in the range 0-31
#define TFT_RST  32//morado // Reset pin

//#define TOUCH_CS 21

#define TFT_WR   19//amarillo  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD   23//naranja

#define TFT_D0   13//marron  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   12//rojo  // so a single register write sets/clears all bits
#define TFT_D2   14//gris
#define TFT_D3   27//morado
#define TFT_D4   26//azul
#define TFT_D5   15//verde
#define TFT_D6   2//amarillo
#define TFT_D7   4//naranja

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
```

Wiring is triple checked and User_Setup_Select.h points to above settings

I have tried with all the available drivers without success.

As mention, I have another 3.5' display, and wire identically but with R61581_DRIVER and works perfectly

Do you know what else I have to ckeck?

TIA

Bodmer, 2021-09-12:
Run the Read_User_Setup diagnostic sketch and check the output is as you expect. Does the output look OK?

adolfonovo, 2021-09-12:
Hi @Bodmer !!
Yes, it is ok. Just found what was happening. It seem, this model needs both 3.3v and 5v .

Now its working. Thanks a lot and congrats!!

closed 2021-09-12 by adolfonovo