Bodmer/TFT_eSPI #1339: SPI bus type ST7789 starts the screen for the first time

SyXavier, 2021-09-16:
Hello Bodmer.Every time I restart after the power is off, the screen will flicker at first, and then the display will be normal.
[image]

Bodmer, 2021-09-16:
I suspect his is because the signal lines are floating (undriven) until the processor has come out of reset and booted. Once the library is invoked the data  lines are configures the GPIO as output. While floating the display sees electrical "noise" pickup and looks like commands and data.

This tends to be more of a problem with displats that do not have a chip select signal.  You might be able to stop this flickering by adding pullups (say 4K7 to 10K range) to at least the SCL line. It is considered good elctronics design practice to avoid floating data/control bus lines.

SyXavier, 2021-09-17:
I tried to add a pull-up 10k to the SCL pin, but it didn't work.

Bodmer, 2021-09-18:
Try a 10K pull down on the RES pin instead.

SyXavier, 2021-09-18:
Adding a 10k pull-down on the res pin still doesn't work.

Bodmer, 2021-09-18:
In that case it seems to be an inherent behaviour of a display without a chip select. The only other option is to switch off the backlight by some means until the display has been initialised by tft.init().

Check the back of the PCB to see if it has a transistor connected to the BLK pin, if so then you can turn on/off the backlight with a GPIO pin.

Bodmer, 2021-09-18:
I have a similar display and it does not exhibit the flicker problem. It does have a backlight transistor:

[image]

Post pictures of your setup and describe the connections. Is the SPI bus shared with other devices?

SyXavier, 2021-09-19:
You and I use the same screen. SPI bus has only screen device.

SyXavier, 2021-09-19:
I have tried tft.init () before closing the backlight, tft.init () after calling tft.fillScreen (), and finally open the backlight. However, when powered on, the blurred screen is displayed first, and then filled with color.

SyXavier, 2021-09-19:
I replaced the st7789_ 2.h, the screen time is shortened, and tft.fillSceen (TFT_BLACK) is called after tft.init (), first showing the black, then the blurred screen.

SyXavier, 2021-09-19:
I suspect that the blurred screen will appear when initializing the TFT, so calling fillsceen () doesn't work.

Bodmer, 2021-11-28:
This appears to be inherent behaiour of the display and I do not see how a change to TFT_eSPI will resolve this.

closed 2021-11-28 by Bodmer