Bodmer/TFT_eSPI #1340: LILY Pi with ili9481: issue+fix

perlix, 2021-09-17:
Using library version 2.3.70 with Setup51_LilyPi_ILI9481.h, my LILYGO LILY Pi showed distorted graphics and wrong colours.

It turned out that none of the current ILI9481_INIT_n sections in TFT_Drivers/ILI9481_Init.h worked for my version of this device. The ILI9481_INIT_1 section has the right commands, but I had to make execution of 'writedata(0x55)' and 'writecommand(TFT_INVON)' unconditional (as a quick fix; adding an ILI9481_INIT_7 section for my version of this device would have been nicer, I suppose).  
 
Now everything works as expected from this great library.

-------------------------------------------------------------------------------
P.S. Setup51_LilyPi_ILI9481.h sets the SPI frequency to 26.666666 MHz, but the display of my LILY Pi still works great (and very fast) at 80 MHz!

Output from TFT_graphicstest_one_lib.ino, 480x320 ili9481, SPI @ 80MHz: 
Screen fill                      158982        
Text                                10746
Lines                              99414
Horiz/Vert Lines              14501
Rectangles (outline)          8063
Rectangles (filled)          384301
Circles (filled)                  58443
Circles (outline)               38007
Triangles (outline)            21496
Triangles (filled)             137050
Rounded rects (outline)   18214
Rounded rects (filled)     392156

Bodmer, 2021-09-18:
OK, I do not have a Lily Pi to test, the setup was provided by a user in pull request #1062.

I am a bit suspicious that you actually have an ST7796 screen as the ILI9481 does not operate above ~30MHz! See #1216 for a setup file.

Let me know what you discover.

Bodmer, 2021-09-18:
See #1341 which has now been merged.

perlix, 2021-09-18:
Hi Bodmer,

The LILY Pi factory firmware (https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/tree/master/examples/LilyPi/lv_demo_widgets) runs fine on my model if I uncomment the #define for ILI9481 in its config.h file. Selecting ST7796S instead produces a blank screen, so I presume that I have the ILI9481 variant. 

TTGO_TWatch_Library contains a copy of TFT_eSPI (version 2.2.15). Copying its ILI9481_Init.h file into version 2.3.70 in my Arduino/libraries folder solved my problems.

So it's a (very welcome) mystery why my LILY Pi runs without any problems @ 80MHz. Could it have some other driver chip that is software compatible with your ILI9481 driver? After all, LILYGO is notorious for releasing their products in many flavours...

perlix, 2021-09-18:
Just tested Setup52_LilyPi_ST7796.h on my LILY Pi. It produces a blank (black) screen, consistent with selecting ST7796S in the config.h file of the earlier mentioned factory firmware sketch. For now, I'll happily stick to Setup51_LilyPi_ILI9481.h and my little workaround. This library is amazing!

perlix, 2021-09-20:
I just confirmed that my LILY Pi has a GT911 touchscreen, a further indication that it really is the ILI9481 version (the ST7796 version has a FT6236). Is it conceivable that it operates at 80 MHz because it's an IPS display?

perlix, 2021-10-16:
Mystery (largely) solved; my mistake.

I use a customized 'User_Setup_Select.h' file, in order to preserve self-created setup files during library updates.  
After an update, I forgot to copy the new "DON'T TINKER WITH ANY OF THE FOLLOWING LINES..." block.
As a result, my file missed the "invoke 18 bit colour for selected displays" section.

Everything works fine now, without any changes in 'ILI9481_Init.h'.
That is to say: even with SPI frequency set to 80MHz in 'Setup51_LilyPi_ILI9481.h'. 
Benchmark results, though, are about half as fast as with LILYGO's version of 'ILI9481_Init.h' and my outdated 'User_Setup_Select.h' file. LILYGO appears to use a 16 bit colour interface: writedata(0x55) instead of writedata(0x66).

closed 2021-10-16 by perlix

Juno63, 2022-03-13:
Hi Perlix,
I am trying to get a LilyPi ILI9481 (GT911) working with the latest TTGO_TWatch_Library V1.4.2, but just uncommenting the #define LILYGO_BLOCK_ILI9481_MODULE` (and commenting #define LILYGO_BLOCK_ST7796S_MODULE) doesn't seem enough to make the lv-demo_widgets example work. Exactly what do I have to change more?

perlix, 2022-04-01:
Hi Juno63,
Sorry, only just noticed your question. The lv-demo_widgets example from the library works on my LilyPi ILI9481 without any further changes.