Bodmer/TFT_eSPI #1347: delay in drawglyph function

SiavashSkynet, 2021-09-23:
https://github.com/Bodmer/TFT_eSPI/blob/c11ecd6932d50cdd63260e346a8b10eb519c7a69/Extensions/Smooth_font.cpp#L401

Hi Bodmer, 

In mentioned function the delay is causing delay in rendering RTL font.
Would you please check it out?

Regards

SiavashSkynet, 2021-10-16:
Update:

I did try to compress the example.vlw font before uploading and use it in my program but the problem is still exists.
Would it be okay if you check it out?

Regards

Bodmer, 2021-11-24:
The deleay is due to the way SPIFFS operates and is not caused by the TFT_eSPI library. The comment is just a reminder and does not indicate a problem with TFT_eSPI.  You may find using LittleFS is a faster or using the font stored in Flash as an aray, see examples.

closed 2021-11-24 by Bodmer