Bodmer/TFT_eSPI #1348: Support for ESP32-S2 + ILI9488 (Wireless-Tag WT32-86)

paultbarrett, 2021-09-24:
Wireless-Tag have released a new LCD/ESP combo which is using an ESP32-S2 (ESP32-S2-WROVER)

This incorporates an ILI9488 connected via parallel however I have been unable to get this to work.

I have mapped out the pins and it appears that the the LCD is connected across 2 registers which I beleive is the reason for it not working. Unfortuantely it isnt possible to change the IO pins used for the parallel interface so wanted to know if there is an alternate option (I suspect it will be a bit slower).

```
#define ESP32_PARALLEL
#define TFT_PARALLEL_8_BIT
#define ILI9488_DRIVER

#define TFT_WIDTH  320
#define TFT_HEIGHT 320

#define USE_HSPI_PORT 

#define TFT_BACKLIGHT_ON HIGH

// ESP32 pins used
#define TFT_CS   -1  // Chip select control pin
#define TFT_DC   38  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  37  // Reset pin

#define TFT_WR    39  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    -1

#define TFT_D0   36  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   35  // so a single register write sets/clears all bits
#define TFT_D2   34
#define TFT_D3   33
#define TFT_D4   21
#define TFT_D5   18
#define TFT_D6   17
#define TFT_D7   16
#define TFT_BL   0
```

FamilieKlein, 2021-10-15:
I also didn't get the display to work with tft_espi.
I have switched to **LovyanGFX** and **LVGL 8**, so I have the display running successfully with the built-in ESP32-S2 and can do pretty much everything with it.

paultbarrett, 2021-10-15:
@FamilieKlein, thanks it looks really promising. I will check it out.

paultbarrett, 2021-10-15:
> […](#)

Thanks, I should be OK, I'm trying to get OpenHASP working on the WT86-32. Will try using this new library in place of TFT_eSpi

Bodmer, 2021-11-18:
The library now supports the ESP32 S2, this has been tested with the Arduino ESP32 board package 2.0.1.

A new setup file example has been added:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70_ESP32_S2_ILI9341.h

closed 2021-11-18 by Bodmer

fvanroie, 2022-01-23:
@paultbarrett I am also trying to get this device working, thanks for sharing your config.

@FamilieKlein Would you mind sharing your working example please?