Bodmer/TFT_eSPI #1352: TFT_eSPI with ESP32-S2 does not compile

homonto, 2021-10-04:
hi, with old boards ESP32 all is ok, but now with ESP32-S2 the code does not compile at all with the errors:

`In file included from .../Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:20:
.../Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:25:34: error: 'VSPI_HOST' was not declared in this scope
     spi_host_device_t spi_host = VSPI_HOST;
                                  ^~~~~~~~~
.../Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:25:34: note: suggested alternative: 'SPI_HOST'
     spi_host_device_t spi_host = VSPI_HOST;
                                  ^~~~~~~~~
                                  SPI_HOST
`

is there any way to solve it?

thank you

nikthefix, 2021-10-05:
For a quick and dirty way to get the latest TFT_eSPI/eFEX to work with the S2 do the following:

Go to TFT_eSPI_ESP32.c in TFT_eSPI lib / processors folder.
Change line 12 HSPI to FSPI
Go to  User_Setup.h
Uncomment #define USE_HSPI_PORT

Go TFT_eFEX.h and change #ifdef ESP32 to ESP32S2 on line 17, 22, 54, 116
Go TFT_eFEX.cpp and change #ifdef ESP32 to ESP32S2 on line 676, 1389

Done!

homonto, 2021-10-05:
> For a quick and dirty way to get the latest TFT_eSPI/eFEX to work with the S2 do the following:
> 
> Go to TFT_eSPI_ESP32.c in TFT_eSPI lib / processors folder. Change line 12 HSPI to FSPI Go to User_Setup.h Uncomment #define USE_HSPI_PORT

DONE

> Go TFT_eFEX.h and change #ifdef ESP32 to ESP32S2 on line 17, 22, 54, 116 Go TFT_eFEX.cpp and change #ifdef ESP32 to ESP32S2 on line 676, 1389

NO SUCH FILES found 
with regards to *h file:
`find . -name TFT_e\*.h

../Arduino2/libraries/TFT_eSPI/Processors/TFT_eSPI_Generic.h
../Arduino2/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP8266.h
../Arduino2/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h
../Arduino2/libraries/TFT_eSPI/Processors/TFT_eSPI_STM32.h
../Arduino2/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h
../Arduino2/libraries/TFT_eSPI/TFT_eSPI.h
../Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_Generic.h
../Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP8266.h
../Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.h
../Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_STM32.h
../Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_RP2040.h
../Arduino/libraries/TFT_eSPI/TFT_eSPI.h`

with regards to *cpp
`find . -name TFT_e\*.cpp

../Arduino2/libraries/TFT_eSPI/TFT_eSPI.cpp
../Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp`

so I am not sure which one
help please ;-)

nikthefix, 2021-10-05:
TFT_eFEX is a supplementary support lib which you may not have installed so you can't find those files. You can download it from Bodmer's github if you need the extra functions. If not then the mods you've done already should be enough to work with the S2.

homonto, 2021-10-05:
you are completely right - modifying the first 2 files was enough to make it working
are we expecting update on the library itself to avoid these changes in the future?

btw: T H A N K  YOUUUUU ;-)

nikthefix, 2021-10-05:
I can't speak for Bodmer but given that he was quite rightly waiting for stable S2 support from the Esp Arduino core, integration will be forthcoming I imagine. BTW, the mods I suggested are just for SPI. If you need parallel interface then further substitution of #ifdef ESP32 for #ifdef ESP32S2 and references of HSPI to FSPI are required but it's the same process. Search and replace cos ESP32-S2 has no VSPI.

homonto, 2021-10-05:
thank you - I have few different displays but all are SPI
I am closing the topic and waiting for update in the main stream

Bodmer, 2021-11-18:
The library now supports the ESP32 S2, this has been tested with the Arduino ESP32 board package 2.0.1.

A new setup file example has been added:
https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70_ESP32_S2_ILI9341.h

homonto, 2021-11-18:
Thank you @Bodmer . The update from 2.0 to 2.0.1: will that kill my previous, manual changes as in the comment above ( did manually HSPI) or everything shall go ok? I am asking as I am not sure if I should "TimeMachine" my laptop before upgrading the library or not :-)

Bodmer, 2021-11-18:
2.0.0 will work with the updated library.

voltar-tech, 2021-11-20:
Thanks, Bodmer... I am a big fan (and user) of the TFT_eSPI library and have successfully ported it to new custom-designed hardware.
Looking forward to support for the forthcoming ESP32-S3 chip one day.  Is there anywhere I can make a financial donation to your overall support efforts?

Bodmer, 2021-11-21:
The changes made for the S2 look compatible with the C3, so it may work.

homonto, 2021-11-21:
> 2.0.0 will work with the updated library.

confirmed - thank you

homonto, 2022-02-03:
latest update of the library ("version": "2.4.32") reopens the issue - cannot compile with ESP32-S2

I restored library "version": "2.4.2" from my backup and I can compile again

Bodmer, 2022-02-03:
It might be this line:

https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L16

This was added to provide compatibility with board package 1.0.6.

homonto, 2022-02-03:
I thought so as well ;-)
but now, what to do?
I saw a lot of new boards in the new version, but what can I do if this version does not compile?

Bodmer, 2022-02-04:
Comment out that line and it should be OK.

I will have to work out a way to solve this.

Bodmer, 2022-02-04:
Closing as #1601 will cover this.

closed 2022-02-04 by Bodmer

johnny49r, 2022-02-06:
TFT-eSPI V2.4.32 & higher doesn't compile with the ESP32-S2 variant.
If I compile with 2.4.31 my code compiles and the display works. If I compile with 2.4.32 or higher I get many compile errors. 
The latest version I have on platformio is 2.4.34. Is there a new version coming that might fix this issue?
Thank you

Bodmer, 2022-02-06:
2.4.2 is the latest released version.

2.4.35 is the latest master copy.

Try the latest and list the errors, I am working away and unable to do any tests for 2 weeks.

Bodmer, 2022-02-06:
See #1601 also.