Bodmer/TFT_eSPI #1358: User config for 1.77' TFT 160(RGB)x128 board

dirkx, 2021-10-09:
```
// Config for the blue boards:
// front: 1.77' TFT 160(RGB)x128
// back:  1.77' TFT and two rows of pins; 8..1 and 9..11
//
// https://www.arthurwiz.com/software-development/177-inch-tft-lcd-display-with-st7735s-on-arduino-mega-2560
// https://www.hackster.io/ocnarf/cheap-1-77-inch-tft-screen-on-esp32-ebe208

#define ST7735_DRIVER     

#define TFT_WIDTH  128
#define TFT_HEIGHT 160

#define ST7735_BLACKTAB
#define TFT_RGB_ORDER TFT_RGB  
#define TFT_INVERSION_OFF

#define TFT_MISO -1  // not broken out on this board.
#define TFT_SCLK 17  // TX2
#define TFT_MOSI 16  // RX2
#define TFT_RST   4  // D4
#define TFT_DC    2  // D2
#define TFT_CS   15  //D15
                     // requires pin 8 (LED ANODE) to be wired to 3V3 or a digital out.

#define TFT_BACKLIGHT_ON HIGH 

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
#define SPI_READ_FREQUENCY  20000000
#define SPI_TOUCH_FREQUENCY  2500000
#endif
```

Bodmer, 2021-10-29:
Thanks, can you provide a clickable link to the display vendors page.

Maybe it is [this one](https://www.aliexpress.com/item/32265119437.html?src=google&src=google&albch=shopping&acnt=631-313-3945&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbActive=false&isSmbAutoCall=false&needSmbHouyi=false&albcp=14261781308&albag=&trgt=&crea=en32265119437&netw=x&device=c&albpg=&albpd=en32265119437&gclid=Cj0KCQjwt-6LBhDlARIsAIPRQcLAFiZH36-DkOG8IDsPIiHD4nNvJQzSPsaLw9x0X7KhGuF6yA_61mQaArWVEALw_wcB&gclsrc=aw.ds&aff_fcid=c5da8ece389245c68d89244dbf19019b-1635545120021-06621-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=c5da8ece389245c68d89244dbf19019b-1635545120021-06621-UneMJZVf&terminal_id=ac1083d2424e4a9388d9bb79c9314268)????

dirkx, 2021-10-30:
The blue one of https://nl.aliexpress.com/item/32265119437.html <https://nl.aliexpress.com/item/32265119437.html> is indeed correct.

Confirmed/tested is https://www.amazon.nl/dp/B07TJGF8HS/ref=twister_B084257CK2 <https://www.amazon.nl/dp/B07TJGF8HS/ref=twister_B084257CK2> as well as https://nl.aliexpress.com/item/32818069069.html <https://nl.aliexpress.com/item/32818069069.html>

With kind regards, in addition to what the TFT_eSPI library offers -   tft.writecommand(onoff ? ST7735_DISPON : ST7735_DISPOFF); on/off for screen saving / preventing burn in also works.