Bodmer/TFT_eSPI #1367: RP2040 Pico-Restouch-LCD-3.5 ILI9488 No video output

penny64, 2021-10-17:
I purchased the waveshare display from https://www.waveshare.com/pico-restouch-lcd-3.5.htm
I'm using PlatformIO with the following config

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:pico]
platform = raspberrypi
board = pico
framework = arduino
lib_deps =
    bodmer/TFT_eSPI#9e64092f58bc4b2458d51ad5bcb0b2e7fdcb6507
    SPI

build_flags =
  -D USER_SETUP_LOADED=1
  -D ILI9488_DRIVER=1                           ; Select ILI9163 driver
  -D TFT_WIDTH=480                              ; Set TFT size
  -D TFT_HEIGHT=320
  -D TFT_MISO=12                                ; Define SPI pins
  -D TFT_MOSI=11
  -D TFT_SCLK=10
  -D TFT_CS=9
  -D TFT_DC=8                                  ; Data/Comand pin
  -D TFT_RST=15                                 ; Reset pin
  -D TOUCH_CS=16
  -D LOAD_GLCD=1                                ; Load Fonts
  -D SPI_FREQUENCY=27000000                     ; Set SPI frequency
  -D SPI_FREQUENCY=70000000`
```
I've tweaked the SPI frequency around, on the library that comes with the device a wide range of frequencies works and none of them work here.
The display does not seem to turn on or do anything with any frequency.

Bodmer, 2021-10-29:
I suspect this is down to the default SPI pins being used and the pins specified being ignored.

Which Arduino RP2040 board package are you using?

Bodmer, 2021-11-28:
Assumed resolved.

closed 2021-11-28 by Bodmer

penny64, 2021-11-29:
Hi I'm using a regular PR2040 directly from the foundation

Bodmer, 2021-11-29:
Run the Read_User_Setup to check the compiler is picking up your settings.

Kabron287, 2021-12-03:
it manages to make reset and then it loops in the file spi.h string 125

penny64, 2021-12-29:
Still broken

Bodmer, 2021-12-29:
The Arduino Mbed package for the RP2040 crashes if SPI port 1 pins are used. This is a bug in the board package.

Earle Philhower's superior RP2040 board package traps the configuration error if SPI port 1 pins are used and does not crash but it will not drive the display. This is because SPI port 0 is the default.

This is a limitation of the board package as both default to SPI port 0 pins use.

If you are not using the SPI port 0 pin numbers from the example setups and are instead using SPI port 1 pins then #define TFT_SPI_PORT 1 must be used in the setup file. This is supported in the latest TFT_eSPI release on Github.