Bodmer/TFT_eSPI #1392: STM32H7xxx - I could not get it working

knopserl, 2021-10-31:
Hi, I really love this library, it's the no 1 TFT lib!
I use it mostly with ESP32, but need to do some projects with STM32H7 boards.
Such as: 
https://github.com/mcauser/MCUDEV_DEVEBOX_H7XX_M
https://de.aliexpress.com/item/1005003019945586.html?spm=a2g0o.productlist.0.0.12513919I8je2A&algo_pvid=500859c3-3ad4-4542-ba56-ee4224c69c8b&algo_exp_id=500859c3-3ad4-4542-ba56-ee4224c69c8b-3&pdp_ext_f=%7B%22sku_id%22%3A%2212000023271084928%22%7D
Schematic:;
https://raw.githubusercontent.com/mcauser/MCUDEV_DEVEBOX_H7XX_M/master/docs/STM32H7XX_M_schematics.pdf

This board has an extra deticated OLED/TFT connector with SPI pinout etc. 
I spent a whole day to find any combination which works with a couple of ILI9431 (320x240), or ILI9488/86 (480x320).
The displays work on e.g. a Teensy 4.0 but not with this SZM32H7 board.
I've used several user setups as template and created a couple of them for the STM32 (I attach one of them). I used the deticated SPI Port connector (see borad PCB) and other SPI ports, Basically SPI1 and SPI2 for this MCU and defined also the STM32, TFT_SPI_PORT (1 and 2 with it's port pins).
I'M using the latest Arduino 1.8.16 IDE with the correct board and MCU type and itÄs flashing correct (tested with the Blink Sketch).
Any idea why it works with all my othe MCU's but not with this board (which is a new working board)?
[User-Setups.zip](https://github.com/Bodmer/TFT_eSPI/files/7449755/User-Setups.zip)

Bodmer, 2021-11-05:
The only STM32H7 board that the library has been tested with is the Nucleo 144 STM32H743ZI2.

You may find the generic processor support will work, to try this comment out the #define:
```
// #define STM32
```
in the setup file, the generic support may however be slow as it uses basic Arduino SPI library calls.

You may need to use the standard Arduino Dxx pin numbers but I am not sure about that.

closed 2021-11-05 by Bodmer

KenjutsuGH, 2021-11-08:
I have the same STM32 board and I also can not get the library with an  ILI9431 (320x240) display.

I was wondering if the OP (@knopserl) was able to get it working yet?