Bodmer/TFT_eSPI #1397: Problem RP2040 + RPI 3.5 display

AndresBellero74, 2021-11-03:
Hi, i am trying to use the raspberry pi pico with the RPI35 display and the image appears but it does not show me quite well in a gray background color instead of blue and diffuse.
These are the connections:
#define TFT_MISO 4
#define TFT_MOSI 3
#define TFT_SCLK 2
#define TFT_CS 1 // Chip select control pin
#define TFT_DC 0 // Data Command control pin
#define TFT_RST 7 // Reset pin (could connect to RST pin)
and these are the uncommented displays:
#define RPI_DISPLAY_TYPE // 20MHz maximum SPI
#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
and attached as it look
[image]
s:

AndresBellero74, 2021-11-03:
[image]
Apply the solution of issue 1213 and now it works but it does not seem to clean the screen well.
Partial fix Raspberry Pi Pico with RPI Display Type #1213.
In the image you can see the error, I am improving and I think we are close, if I am missing something please let me know.
Thank you so much.
Regards..

Bodmer, 2021-11-05:
This should now be fixed in the latest version 2.3.73.

Try it and post back if you have a problem.

There were two problems:
1. drawPixel needed to be updated.
2. The DC and CS lines were being toggled too quickly by the RP2040 so I have slowed then down.

My display runs at 32MHz but typically these only run up to 20MHz SPI clock rate.

closed 2021-11-05 by Bodmer

AndresBellero74, 2021-11-05:
Hi, how are you? Thank you for the answer. I need to know where the 2.3.73 version is, since the version that appears as the latest 2.3.70 directly does not show me anything, only a blank screen.
Regards.

AndresBellero74, 2021-11-05:
[TFT_eSPI-master.old.zip](https://github.com/Bodmer/TFT_eSPI/files/7479838/TFT_eSPI-master.old.zip)
I'll give you the library and in "example bellero" is the simpler example than with 2.3.70, which is the last one I found, nothing comes out, the screen is blank.

Bodmer, 2021-11-05:
The update has not been released yet but can be downloaded from the main repository master:

[image]

AndresBellero74, 2021-11-05:
Thank you very much for the answer, it worked perfectly, very grateful, I give you the video of how the cube demo works now.
Ask if you have something already done to show images that I can test, the idea is to reproduce bmp or jpg with the raspberry pi pico.
a total genius and very good response time.

https://user-images.githubusercontent.com/73205398/140504968-63f93b10-423b-494b-aba1-b287899e5a65.mp4

Bodmer, 2021-11-05:
The simplest way to show images is to store them as an array in FLASH. I have not tried to connect an SD card to the RP2040, but that is an option if you have many files.

Try the Flash_Jpg_DMA example with line 7 commented out thus:
```
//#define USE_DMA
```

Look at the panda.h tab in that sketch to see how jpegs can be converted to arrays.

You will need to install the TJpg_Decoder Arduino library from the library manager or download [from here](https://github.com/Bodmer/TJpg_Decoder).

AndresBellero74, 2021-11-10:
Thanks to your help and the author of the SD library for the raspberry pico I was able to play BMP files with the SD and on the 3.5 "display of the raspberry pi.
I'll pass it to you in case anyone is interested.
This is the project link:
https://github.com/AndresBellero74/Raspberry-PI-PICO-display-RPI35
Regards.