Bodmer/TFT_eSPI #1401: SSD1963 ESP32 no backlight with PWM pin controlling VLed

hannes314, 2021-11-05:
Hello,

i am trying to get a 7" LCD from MikroElektronika to work. It is based on the SSD1963. I am using an esp32 wroom with 38 pins.
The connector has no separate pin for the backlight and after editing the User_Setup_Select.h to use the SSD1963_800_driver the display stays off. So i looked up the schematic for the board and the backlight is controlled by the PWM pin of the driver but the pin stays low after power is supplied. 

Schematic for the display:
https://download.mikroe.com/documents/tft-board-capacitive/tft-board-7-capacitive-schematic-v106.pdf

The datasheet of the driver states that the driver has dynamic backlight control (DBC). The registers for DBC and PWM are set by the library as followed:

DBC:
[image]

PWM_Conf:
[image]

In Summary:
DBC is enabled with manual brightness.
PWM is enabled with some frequency(?), 50% duty cycle which is ignored due to the enabled DBC, PWM is controlled by Host (?), x240/255 brightness and minimum duty cycle of 0%

I tried disabling the DBC by changing the 0xd0 register in the SSD1963_init.h file at the 800 driver from 0x0d to 0x0c with no effect.
I also tried to set the minimum brightness value to a high value with no effect.
I double checked the wiring.
Sometimes when I press the enable button of the esp the backlight flashes white but stays off after that.

At this point I do not know what the issue could be.
Any help would be appreciated.

Bodmer, 2021-11-05:
I have not used that feature... I think you are on the right track and are trying exactly what I would try.

Is there any example code from Mikroe that you can examine?

Bodmer, 2021-11-05:
There is some [code here ](http://www.ampdisplay.com/documents/pdf/SSD1963%20800480S.txt)that may give you a clue. It configures the GPIO clocks and the DBC registers.

Bodmer, 2021-11-05:
I also found this init code on the [BuyDisplay](https://www.buydisplay.com/7-tft-screen-touch-lcd-display-module-w-ssd1963-controller-board-mcu) website (that is the display I have). Part of this code configures the GPIO, clocks and sets the PWM for the backlight.  I did not need that code as my display can be configured to control the backlight with an external pin.
```
void SSD1963_Initial(void)
{
    
  /* 
	reset=0;
	delay(100);
	reset=1;
	delay(100);*/
	write_command(0x00E2);	//PLL multiplier, set PLL clock to 120M
	Write_Data_int(0x0023);	    //N=0x36 for 6.5M, 0x23 for 10M crystal
	Write_Data_int(0x0002);
	Write_Data_int(0x0054);
	write_command(0x00E0);  // PLL enable
	Write_Data_int(0x0001);
	delayus(10);
	write_command(0x00E0);
	Write_Data_int(0x0003);		// now, use PLL output as system clock
	delayus(10);
	write_command(0x0001);  // software reset
	delayus(20);
	write_command(0x00E6);	//PLL setting for PCLK, depends on resolution
	Write_Data_int(0x0003);
	Write_Data_int(0x0033);
	Write_Data_int(0x0033);

	write_command(0x00B0);	//LCD SPECIFICATION
	Write_Data_int(0x0020); //24 bit TFT panel
	Write_Data_int(0x0000); //Hsync+Vsync +DE mode  TFT mode
	Write_Data_int((799>>8));  //Set HDP
	Write_Data_int(799);
    Write_Data_int(479>>8);  //Set VDP
	Write_Data_int(479);
    Write_Data_int(0x0000);

	write_command(0x00B4);	//HSYNC
	Write_Data_int(0x04);  //Set HT
	Write_Data_int(0x1f);
	Write_Data_int(0x00);  //Set HPS
	Write_Data_int(0xd2);
	Write_Data_int(0x00);			   //Set HPW
	Write_Data_int(0x00);  //Set HPS
	Write_Data_int(0x00);
	Write_Data_int(0x00);

	write_command(0x00B6);	//VSYNC
	Write_Data_int(0x02);   //Set VT
	Write_Data_int(0x0c);
	Write_Data_int(0x00);  //Set VPS
	Write_Data_int(0x22);
	Write_Data_int(0x00);		//Set VPW
	Write_Data_int(0x00);  //Set FPS
	Write_Data_int(0x00);

	write_command(0x00B8);
	Write_Data_int(0x000f);    //GPIO is controlled by host GPIO[3:0]=output   GPIO[0]=1  LCD ON  GPIO[0]=1  LCD OFF 
	Write_Data_int(0x0001);    //GPIO0 normal

	write_command(0x00BA);
	Write_Data_int(0x0001);    //GPIO[0] out 1 --- LCD display on/off control PIN

	write_command(0x0036); //rotation
	Write_Data_int(0x0008);//RGB=BGR

	write_command(0x003A); //Set the current pixel format for RGB image data
	Write_Data_int(0x0050);//16-bit/pixel

	write_command(0x00F0); //Pixel Data Interface Format
	Write_Data_int(0x0003);//16-bit(565 format) data 

	write_command(0x00BC); 
	Write_Data_int(0x0040);//contrast value
	Write_Data_int(0x0080);//brightness value
	Write_Data_int(0x0040);//saturation value
	Write_Data_int(0x0001);//Post Processor Enable

	delayus(5);

	write_command(0x0029); //display on

	write_command(0x00BE); //set PWM for B/L
	Write_Data_int(0x0006);
	Write_Data_int(0x0080);
	Write_Data_int(0x0001);
	Write_Data_int(0x00f0);
	Write_Data_int(0x0000);
	Write_Data_int(0x0000);

	write_command(0x00d0); 
	Write_Data_int(0x000d);

}
```

hannes314, 2021-11-06:
Thank you for your help.
I tried your examples of setting the registers with no success. So I manually provided the backlight with some voltage and noticed that also the the lcd does not work. After some time there appear random colored vertical lines on the white screen and restarting the esp has no effect at all. That is why I tried the TFT_ReadWrite_test from the example. No errors there. Do you have any ideas why the driver is communicating with the esp but nothing is showing on screen? 
I noticed from the schematic of the board that the lcd connector has a mode pin which is grounded by a jumper. This is different from lcd used in [this thread](https://github.com/Bodmer/TFT_eSPI/issues/704) where the pin is tied to 3.3V.
I tried both jumper settings without luck. Do you know of any other hardware related settings that could be wrong?

hannes314, 2021-11-06:
I probed the LCD-CLK, VSYNC and HSYNC pins on the lcd connector. There is just 70mV noisy DC on there. This is not how it is supposed to be, is it?

hannes314, 2021-11-06:
So I looked up some more settings to see if they match the hardware and in the SSD1963_init.h file the PLL multiplier for the SSD193_800_DRIVER is not set to match the external 10M clock. After changing those registers according to how the other drivers are initialized, the Colour_Test example is kinda working. The Text is not in the upper left corner but shifted a little lower and some letters have a partially wrong color. After inverting two times the display goes to black and stays off. The other examples just produce a scrambled flash of pixels at startup and then the display stays off. Could this still be an error based on the settings in the registers written in the init.h file?

hannes314, 2021-11-07:
Okay, now the display is back to not working again. That is why I googled again and found a [thread](https://forum.arduino.cc/t/buydisplay-4-3-tft-lcd-screen-doesnt-do-anything-ssd1963-tft-espi-with-esp32/886873/15)  on the arduino forum where someone has the same issues as me. The SSD1963 responds to writing and reading pixel data but will not display anything. It looks like it fails to initialize. In the forum the solution was to power the esp from various sources so I tried a wall adapter and connected esp gnd to lcd gnd since the adapter is isolated and now it is working again. I was powering the lcd from an ATX power supply and the esp from my computer without connecting the grounds. Stupid mistake.
Now the X0 and Y0 coordinates are still not in the upper left corner. Is this a setting in the Vsync and Hsync registers? I found an example initialization from a mikroe library so I think I can figure that out.

closed 2021-11-18 by Bodmer