Bodmer/TFT_eSPI #1408: When compiling error:  'class TFT_eSPI' has no member named 'getTouch'

Nulista, 2021-11-07:
Hello, i get this error by compiling this example....

D:\Arduino\libraries\TFT_eSPI-master\examples\480 x 320\Keypad_480x320\Keypad_480x320.ino: In function 'void loop()':
Keypad_480x320:114:22: error: 'class TFT_eSPI' has no member named 'getTouch'
   bool pressed = tft.getTouch(&t_x, &t_y);

Thx
Ludo

brian-not-the-messiah, 2021-11-11:
You haven't defined `Touch_CS` in User_setup... 

if the touch CS is connected to pin 21, you can simply uncomment `#define TOUCH_CS 21`

Nulista, 2021-11-12:
Hello, bodmer,

Yes indeed, after updating the library I have to adjust those things 
too...... stupid of me, sorry.
Thanks for the quick response!

thx Ludo

------ Origineel bericht ------

has no member named 'getTouch' (Issue #1408)

brian-not-the-messiah, 2021-11-12:
No problem! This isn't Bodmer by the way!

Bodmer, 2021-11-18:
Glad this is solved.

closed 2021-11-18 by Bodmer