Bodmer/TFT_eSPI #1416: TFT_eSPI with SD.h / JPEGDecoder on STM32

RicardoCSR, 2021-11-10:
Hi Bodmer, I have been working in a project that use a STM32F401CCU6 with ST7789 (240-240), "future BME280 SPI" and a  MicroSD module, using the TFT_eSPI library and the JPEGDecoder and the arduino compile configured by stm32duino in preferences. I was capable to make the ST7789 runs in STM32 and show some images that could fit in the memory of Micro-controller but the memory was not suffice, i never had work with SPI much less with a tft display and it was my first attempt following some examples and tips.
After that i made the code and config the SD module but when it is start show a half blank screen with lines, after spend a time i understand that the SD module did not start properly i made many attemps with none result. 

Would you help me to solve this problem?

That is my Pins config on STM32F401CCU6

ST7789 with 3.3V 
PA1     Pin DC ST7789
PA2     Pin RST ST7789
PA3     Pin BLK ST7789
PA5     Pin SCL ST7789 
PA7     Pin SDA ST7789

SD Card with 5V 
PA15    Pin CS SD-Card
PB3     Pin SCL SD-Card
PB4     Pin MISO SD-Card
PB5     Pin MOSI SD-Card   

Library's 
#include "STM32LowPower.h"
#include <STM32RTC.h>

#include <TFT_eSPI.h> 
#include <SPI.h>
#include <Wire.h>
#include "Adafruit_GFX.h"

#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>

#include <SD.h>
#include <JPEGDecoder.h>

And the void setup, if I start the SD.begin first the display works but the SD still fails. In example shown below both did not work with that screen.

    Serial.begin(250000);
    analogWrite (dspPinPower, 0);
    LowPower.begin();
    tft.init();
    tft.fillScreen(black);
    Serial.print("Initializing SD Card...");
    if (!SD.begin(SD_PIN)) {
        Serial.println("SD Card Failed");
        digitalWrite(nixie_1, HIGH);
    }
    Serial.println("SD Card Opened!");
    digitalWrite(nixie_1, LOW);
    Serial.println("initialisation done.");

And thank you for made this library that made easy to create for everyone and configure the many possibles displays combined by the speed of rendering, sorry by any orthographic mistake, English is not my first language.

Bodmer, 2021-11-18:
Run the SD library examples first before using the TFT_eSPI examples and check the SD card can be read OK. You will need to connect the TFT_CS pin from the display to +3.3V temporarily (or drive that pin high at the start of setup() ) to run the SD card library examples so the TFT does not drive the SPI bus.

Bodmer, 2021-11-27:
Assumed resolved.

closed 2021-11-27 by Bodmer