Bodmer/TFT_eSPI #1419: Strange readings from touch screen

brian-not-the-messiah, 2021-11-11:
Hi, I am getting some strange reading from my touch screen and I was hoping someone could point me in the right direction... I've tried to get everything I think is relevant together below:

I have a 2.8" ILI9341.
The board definitely has touch, there is an extra ribbon on the front and an extra chip on the back. Here is the product page of the one I bought: https://www.amazon.co.uk/gp/product/B0749N3S33/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

This is connected to a ESP32 v4

The wiring:
[image]

Readings from touch_diagnose:

These are untouched:
```19:42:49.529 -> x: 1472     y: 2672     z: 4096 
19:42:49.779 -> x: 1298     y: 2656     z: 4097 
19:42:50.035 -> x: 1456     y: 2167     z: 4095 
19:42:50.290 -> x: 1205     y: 2664     z: 4095 
19:42:50.545 -> x: 1324     y: 2341     z: 4095 
19:42:50.761 -> x: 1472     y: 2723     z: 4098 ```
These are touched in the middle:
```19:42:51.018 -> x: 1309     y: 0     z: 4095 
19:42:51.267 -> x: 1648     y: 0     z: 4095 
19:42:51.518 -> x: 1690     y: 0     z: 4083 
19:42:51.774 -> x: 1791     y: 0     z: 4095 
19:42:52.027 -> x: 1576     y: 0     z: 4095 
19:42:52.283 -> x: 1807     y: 0     z: 4098```

Here is the read_user_setup:

```
19:48:02.862 -> [code]
19:48:02.862 -> TFT_eSPI ver = 2.3.70
19:48:02.862 -> Processor    = ESP32
19:48:02.862 -> Frequency    = 240MHz
19:48:02.862 -> Transactions = Yes
19:48:02.862 -> Interface    = SPI
19:48:02.862 -> Display driver = 9341
19:48:02.862 -> Display width  = 240
19:48:02.862 -> Display height = 320
19:48:02.862 -> 
19:48:02.862 -> MOSI    = GPIO 23
19:48:02.862 -> MISO    = GPIO 19
19:48:02.862 -> SCK     = GPIO 18
19:48:02.862 -> TFT_CS   = GPIO 15
19:48:02.862 -> TFT_DC   = GPIO 2
19:48:02.862 -> TFT_RST  = GPIO 4
19:48:02.862 -> TOUCH_CS = GPIO 21
19:48:02.862 -> 
19:48:02.862 -> Font GLCD   loaded
19:48:02.862 -> Font 2      loaded
19:48:02.862 -> Font 4      loaded
19:48:02.862 -> Font 6      loaded
19:48:02.862 -> Font 7      loaded
19:48:02.862 -> Font 8      loaded
19:48:02.862 -> Smooth font enabled
19:48:02.862 -> 
19:48:02.862 -> Display SPI frequency = 40.00
19:48:02.862 -> Touch SPI frequency   = 1.00
19:48:02.898 -> [/code]
```

daxliniere, 2021-11-14:
I'm getting a similar issue with my ILI9488 board (https://forum.arduino.cc/t/ili9488-calibration-with-esp32/925095/4)

bsperryn, 2021-11-15:
Turned out the touch panel was faulty. I ordered another and it works without having to make any changes.

brian-not-the-messiah, 2021-11-15:
Turned out the touch panel was faulty. I ordered another and it works without having to make any changes.

closed 2021-11-15 by brian-not-the-messiah

daxliniere, 2021-11-15:
Hey Brian, how did you work out the touch panel was damaged?

brian-not-the-messiah, 2021-11-15:
There was a small crack and it was giving strange values, so I thought no harm in buying another to test whether that resolved the issue.