Bodmer/TFT_eSPI #1420: 4.3 inch LCD 480x272 SSD1963 not working with ESP32 using TFT_eSPI library

tinhvega, 2021-11-12:
I am programming a 4.3 inch 480x272 SSD 1963 with ESP32 using TFT_eSPI library.
I fixed the library as per the instructions:
- #define SSD1963_480_DRIVER
- #include <User_Setups/Setup50_SSD1963_Parallel.h>
- #define SSD1963_480_DRIVER // 272 x 480 display
// ESP32 pins used
#define TFT_CS 33 // Chip select control pin
#define TFT_DC 15
#define TFT_RST 32 // Reset battery

#define TFT_WR 4
#define TFT_RD 2

#define TFT_D0 12
#define TFT_D1 13 // so a single register write sets/clears all bits
#define TFT_D2 26
#define TFT_D3 25
#define TFT_D4 17
#define TFT_D5 16
#define TFT_D6 27
#define TFT_D7 14

I have checked the wire connection, the ESP32 is running but the LCD is still not working.
Please help me. thanks so mu
[image]
[image]
ch

IanBUK, 2021-11-14:
One quick thing to check that's caught me out - ensure that you're turning the backlight on.

tinhvega, 2021-11-15:
I have connected LED pin of LCD to 3.3V. On the picture you can see that the light is on

tinhvega, 2021-11-16:
I was able to fix it.
Connect the STBYB (Standby) pin to high.
All solved

closed 2021-11-16 by tinhvega