Bodmer/TFT_eSPI #1421: Animated dial example corrupting new animating

ppamorim, 2021-11-14:
Hi,

I am testing the library and I found a buffer corruption (is it?) when the animation of the needle is happening, take a look after a minute:

[image]

Note that only the content where the needles passes by gets corrupted. The rest (for example the numbers and chars) is not getting corrupted.

This is the only example that I could cause it to corrupt, other animations are working fine.

Bodmer, 2021-11-18:
This animation requires that the TFT screen can be read and that MISO is wired up.

If the screen is wired up correctly, it is readable and the read SPI rate is OK for the display then it should pass the read/write tests in the example diagnostic sketch TFT_ReadWrite_Test.

ppamorim, 2021-11-21:
@Bodmer You can see my setup [here](https://github.com/ppamorim/ESP32-ST7789V). I will run the test and paste the results here too.

Bodmer, 2021-11-21:
Try reducing the read SPI frequency to 6MHz.
```
#define SPI_READ_FREQUENCY  6000000
```

closed 2021-11-27 by Bodmer