Bodmer/TFT_eSPI #1422: XPT2046 touch screen fails when SPI.begin() is used

Mark-FS, 2021-11-14:
I am trying to set up an ili9341 TFT with XPT2046 touch screen and an RC522 RFID reader on an ESP32 Dev Kit C v4.

The TFT and Touch screen are on HSPI (GPIO13 (MOSI), GPIO12(MISO), GPIO14(CLK) and GPIO15 (CS) GPIO 26 (touch CS)).  The RC522 is on VSPI (GPIO23 (MOSI), GPIO19(MISO), GPIO18(CLK) and GPIO5 (CS)).

The TFT/Touch works fine separately as does the RC522 RFID reader but put them together and the touch fails.  tft.getTouch(&x, &y) always returns false and so never responds to the touch with new x. y values.  If I comment out the SPI.begin() the RC522 stops working but the touch function returns.  The RC522 is running from library MFRC522.h.

Any ideas as to what I might be missing or not understanding?  Thank you...

#include "FS.h"
 
#include <SPI.h>
#include <MFRC522.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

#define CALIBRATION_FILE "/TouchCalData3"
#define REPEAT_CAL false

#define SS_PIN 21
#define RST_PIN 22
MFRC522 mfrc522(SS_PIN, RST_PIN);   // Create MFRC522 instance.

bool cardRead = 0;
bool tftRefresh = 1;

// Comment out to stop drawing black spots
#define BLACK_SPOT
  
void setup() 
{
  tft.init();
  tft.fillScreen(TFT_BLACK);
  tft.setRotation(3);
  
  touch_calibrate();
  
  Serial.begin(115200);   // Initiate a serial communication
//  SPI.begin();      // Initiate  SPI bus
  mfrc522.PCD_Init();   // Initiate MFRC522
  delay(100);
  Serial.println("Approximate your card to the reader...");
  Serial.println();
}

void readRFID(){
  
   if (mfrc522.PICC_IsNewCardPresent()) { // (true, if RFID tag/card is present ) PICC = Proximity Integrated Circuit Card
    if(mfrc522.PICC_ReadCardSerial()) { // true, if RFID tag/card was read
       
  cardRead = 1;
  Serial.print("UID tag :");
  String content= "";

  for (byte i = 0; i < mfrc522.uid.size; i++) 
  {
//     Serial.print(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " ");
//     Serial.print(mfrc522.uid.uidByte[i], HEX);
     content.concat(String(mfrc522.uid.uidByte[i] < 0x10 ? " 0" : " "));
     content.concat(String(mfrc522.uid.uidByte[i], HEX));
  }
  
  tft.setTextSize(2);
  tft.setCursor(10, 70);
  tft.print("RFID tag is: ");
  
  Serial.println("");  
  Serial.println("Message : ");
 
  content.toUpperCase();
   
  tft.setCursor(10, 100);
  tft.println(content);
    
  if (content.substring(1) == "3D A5 D3 62") //change here the UID of the card/cards that you want to give access
  {
    Serial.println("Authorized access");
    Serial.println();
  }
 
 else   {
    Serial.println(" Access denied");
      }
    }
 }
}

void touchCheck()
{
  uint16_t x, y;

  //Serial.println("Checking for touch...");

 if (tft.getTouch(&x, &y)) // See if there's any touch data for us
 {
    Serial.println("Found Data");
    #ifdef BLACK_SPOT    // Draw a block spot to show where touch was calculated to be
    tft.fillCircle(x, y, 2, TFT_GREEN);
    #endif

        Serial.print("Touched at: X");
        Serial.print(x);
        Serial.print(", Y ");
        Serial.println(y);
        
    if (x > 20 && x < 75) {
        if (y > 165 && y < 220) {

        tft.setCursor(10, 130);
        tft.print("0.5 Hour pressed");
        }
    }
 } 

  Serial.print("X:");
  Serial.print(x);
  Serial.print("    Y:");
  Serial.println(y);
 
}

void touch_calibrate()
{
  uint16_t calData[5];
  uint8_t calDataOK = 0;

  // check file system exists
  if (!SPIFFS.begin()) {
    Serial.println("Formating file system");
    SPIFFS.format();
    SPIFFS.begin();
  }

  // check if calibration file exists and size is correct
  if (SPIFFS.exists(CALIBRATION_FILE)) {
    if (REPEAT_CAL)
    {
      // Delete if we want to re-calibrate
      SPIFFS.remove(CALIBRATION_FILE);
    }
    else
    {
      File f = SPIFFS.open(CALIBRATION_FILE, "r");
      if (f) {
        if (f.readBytes((char *)calData, 14) == 14)
          calDataOK = 1;
        f.close();
      }
    }
  }

  if (calDataOK && !REPEAT_CAL) {
    // calibration data valid
    tft.setTouch(calData);
  } else {
    // data not valid so recalibrate
    tft.fillScreen(TFT_BLACK);
    tft.setCursor(20, 0);
    tft.setTextFont(2);
    tft.setTextSize(1);
    tft.setTextColor(TFT_WHITE, TFT_BLACK);

    tft.println("Touch corners as indicated");

    tft.setTextFont(1);
    tft.println();

    if (REPEAT_CAL) {
      tft.setTextColor(TFT_RED, TFT_BLACK);
      tft.println("Set REPEAT_CAL to false to stop this running again!");
    }

    tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);

    tft.setTextColor(TFT_GREEN, TFT_BLACK);
    tft.println("Calibration complete!");

    // store data
    File f = SPIFFS.open(CALIBRATION_FILE, "w");
    if (f) {
      f.write((const unsigned char *)calData, 14);
      f.close();
    }
  }
    tft.fillScreen(TFT_BLACK); 
}

void drawTFT(){

  tft.setCursor(10, 20);    // X, Y
  tft.setTextColor(TFT_WHITE,TFT_BLACK);    // Set the font colour to be white with a black background, set text size multiplier to 2
  tft.setTextSize(1);
  tft.println("OK! Let's read some RFID tags...");
  tft.setCursor(10, 40);
  tft.println("Present RFID card....");

  
//  fillRoundRect(int32_t x, int32_t y, int32_t w, int32_t h, int32_t r, uint32_t color) 
  tft.fillRoundRect(20, 165, 55, 55, 10, TFT_YELLOW); 
  tft.fillRoundRect(95, 165, 55, 55, 10, TFT_YELLOW); 
  tft.fillRoundRect(165, 165, 55, 55, 10, TFT_YELLOW); 
  tft.fillRoundRect(240, 165, 55, 55, 10, TFT_RED); 

  tft.setTextSize(2);
  tft.setCursor(31, 178);    // X, Y
  tft.setTextColor(TFT_BLACK);    // Set the font colour to be white with a black background, set text size multiplier to 2
  tft.println("0.5");

  tft.setTextSize(1);
  tft.setCursor(33, 200);    // X, Y
  tft.setTextColor(TFT_BLACK);    // Set the font colour to be white with a black background, set text size multiplier to 2
  tft.println("Hours");
  
  tft.setTextSize(2);
  tft.setCursor(106, 178);    // X, Y
  tft.setTextColor(TFT_BLACK);    // Set the font colour to be white with a black background, set text size multiplier to 2
  tft.println("1.0");

  tft.setTextSize(1);
  tft.setCursor(109, 200);    // X, Y
  tft.setTextColor(TFT_BLACK);    // Set the font colour to be white with a black background, set text size multiplier to 2
  tft.println("Hours");

  tft.setTextSize(2);
  tft.setCursor(176, 178);    // X, Y
  tft.setTextColor(TFT_BLACK);    // Set the font colour to be white with a black background, set text size multiplier to 2
  tft.println("2.0");

  tft.setTextSize(1);
  tft.setCursor(179, 200);    // X, Y
  tft.setTextColor(TFT_BLACK);    // Set the font colour to be white with a black background, set text size multiplier to 2
  tft.println("Hours");

  
  tft.setTextSize(2);
  tft.setCursor(245, 184);    // X, Y
  tft.setTextColor(TFT_BLACK);    // Set the font colour to be white with a black background, set text size multiplier to 2
  tft.println("STOP");

  tft.setTextColor(TFT_WHITE);

  tftRefresh = 0;

}

void loop() 
{
  if (tftRefresh){
    drawTFT();
  }
  readRFID();
  touchCheck();

  if (cardRead){
  delay(2000);
  cardRead = 0;
  tftRefresh = 1;
  tft.fillScreen(TFT_BLACK);
  }

}

Mark-FS, 2021-11-18:
All sorted.  If it helps someone else the noob mistake was not reading enough.  Right at the bottom of User_Setup is the solution, if you are using the HSPI port on an ESP32 you have to uncomment the line #define USE_HSPI_PORT.  Do that and it works great.  Good luck all...

Bodmer, 2021-11-18:
Thanks for finding the solution!

closed 2021-11-18 by Bodmer

TDVSVI, 2022-11-22:
> All sorted. If it helps someone else the noob mistake was not reading enough. Right at the bottom of User_Setup is the solution, if you are using the HSPI port on an ESP32 you have to uncomment the line #define USE_HSPI_PORT. Do that and it works great. Good luck all...

Can you give me a code? I found problem like you but not solve it. thank you :)