Bodmer/TFT_eSPI #1428: ST7789 TFT LCD working on ESP32 WROOM Dev Kit but not working on ESP32-C3-WROOM-02 Dev Kit in Arduino

cgandrewdev, 2021-11-16:
Hi I have been successfully using the TFT_eSPI library on an ESP32 WROOM dev kit in conjunction with an ST7789 based SPI TFT LCD 240x240 module in the Arduino UI. The library works really well, thank you. 

I have switched to a newer dev kit, the ESP32-C3-WROOM-02 dev kit which uses the newer C3 module. But I have not been able to compile the example project, for example the Read_User_Setup project. It compiles and runs fine when I set the Board to ESP32 Dev Module but as soon as I change the board to ESP32C3 Dev Module and change the port and recompile, I get pages full of errors:

```
In file included from C:\Users\P50\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:20:
C:\Users\P50\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:23:34: error: 'HSPI_HOST' was not declared in this scope
     spi_host_device_t spi_host = HSPI_HOST;
                                  ^~~~~~~~~
C:\Users\P50\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:23:34: note: suggested alternative: 'SPI3_HOST'
     spi_host_device_t spi_host = HSPI_HOST;
                                  ^~~~~~~~~
                                  SPI3_HOST
In file included from C:\Users\P50\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:12,
                 from C:\Users\P50\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.h:65,
                 from C:\Users\P50\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:17:
C:\Users\P50\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/spi_reg.h:23:34: error: 'REG_SPI_BASE' was not declared in this scope
 #define SPI_CMD_REG(i)          (REG_SPI_BASE(i) + 0x0)
                                  ^~~~~~~~~~~~
C:\Users\P50\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:31:60: note: in expansion of macro 'SPI_CMD_REG'
   volatile uint32_t* _spi_cmd       = (volatile uint32_t*)(SPI_CMD_REG(SPI_PORT));
                                                            ^~~~~~~~~~~
C:\Users\P50\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1/tools/sdk/esp32c3/include/soc/esp32c3/include/soc/spi_reg.h:23:34: note: suggested alternative: 'REG_I2S_BASE'
 #define SPI_CMD_REG(i)          (REG_SPI_BASE(i) + 0x0)
                                  ^~~~~~~~~~~~
C:\Users\P50\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_ESP32.c:31:60: note: in expansion of macro 'SPI_CMD_REG'
   volatile uint32_t* _spi_cmd       = (volatile uint32_t*)(SPI_CMD_REG(SPI_PORT));
```

... and so on, I would have attached the full output as a file but I might not be able to add files.

I am not sure where to start, the errors seem to be SPI related.  I have been looking at the specifications for the C3 module and found that it has three SPI ports SPI0, SPI1, and SPI2, only SPI2 is configurable as general purpose SPI. I dont really know what I should be doing to solve this. 

If anyone has any advice it would be much appreciated! Thanks

Bodmer, 2021-11-17:
The new ESP32-C3 is not currently supported. Today I received an ESP32-S2 board so will be working on adapting the library for that processor, maybe the C3 will then work too.

As you note the issue seems to be SPI port references.

cgandrewdev, 2021-11-18:
Thanks very much! I'll be happy to do testing when ready.

Bodmer, 2021-11-18:
The library has been updated for the ESP32 S2 processor variant, the changes may also work with the C3 but this is untested and I do not have a C3 board.

BeijingUncle, 2021-11-19:
There is a post from Twitter maybe useful: https://twitter.com/martinfasani/status/1386019656016728068?s=20

cgandrewdev, 2021-11-30:
Thanks for doing the work on the S2.

I have finally been able to get some time to look at this. I noted that ESP32 Arduino board package 2.0.0 is required per the notes, so I reinstalled with GIT etc. per the instructions here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

I have Arduino 1.8.16, 

The first test, with no dev module plugged in, was to set the target board to ESP32S2 Dev Module and just do a compile, and that worked just fine. 

Then I switched to ESP32C3 Dev Module and tried to recompile but I get a similar list of errors as before. This is without any dev module plugged in at all, just compiling for the C3 dev module as the target. You mentioned that you didnt have a C3 dev module but errors I am getting exist without even a dev module plugged in, just by changing to the C3 module as target.

I have attached the error list if that helps any.

If I selected S2 Dev Module and compiled and tried to upload to the C3 module anyway it wont because it knows I have a C3 Module. I am not sure what else I can do to from here. Thoughts?

[TFT_eSPI errors.txt](https://github.com/Bodmer/TFT_eSPI/files/7624464/TFT_eSPI.errors.txt)

freemanzwin, 2021-12-01:
arduino esp32c3 spi is not implemented yet.
-----------------------------------------------------

esp32s2 spi_reg.h
---------------------------------------------------------------------------------------------------------------

#include "soc.h"
**#define REG_SPI_BASE(i)     (DR_REG_SPI2_BASE + (((i)>3) ? (((i-2)* 0x1000) + 0x10000) : ((i - 2)* 0x1000 )))**

#define SPI_CMD_REG(i)          (REG_SPI_BASE(i) + 0x000)

===================================================================

esp32c2 spi_reg.h
---------------------------------------------------------------------------------------------------------------

#include "soc.h"

#define SPI_CMD_REG(i)          (REG_SPI_BASE(i) + 0x0)

GYSS1204, 2021-12-16:
> The compile problem is caused by bugs and inconsistencies in the IDF for the ESP32C3.
> 
> I have [added some patches](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_ESP32.h#L15-L30) to the latest Github master.
> 
> The examples compile OK now but I have not tested the library on the ESP32C3 as I do not have one!

Hello, I use your latest library to run on ESP32C3 and drive the ST7789 display screen, but it does not work properly. I use oscilloscope to measure the pin and find no SPI signal output.

Bodmer, 2021-12-16:
Thanks for your information. I think we can say the C3 is currently unsupported. I may get a C3 at some point to figure out what is happening.

gabrielcsapo, 2022-01-22:
I recently got the c3 working with https://github.com/PaulStoffregen/XPT2046_Touchscreen. I published an article about it https://www.gabrielcsapo.com/arduino-esp32-c3-32s-ili9341-with-touchscreen/. My next step is getting this library to work with it. @Bodmer if you need a test chip DM me and I can send you one.

gabrielcsapo, 2022-01-24:
I can get the touchscreen to respond, but no dice on any TFT output. 

```
[code]
TFT_eSPI ver = 2.4.2
Processor = ESP32
Frequency = 80MHz
Transactions = Yes
Interface = SPI
Display driver = 9341
Display width = 240
Display height = 320
MOSI = GPIO 6
MISO = GPIO 5
SCK = GPIO 4
TFT_CS = GPIO 7
TFT_DC = GPIO 8
TFT_RST = GPIO 1
TOUCH_CS = GPIO 2
Font GLCD loaded
Font 2 loaded
Font 4 loaded
Font 6 loaded
Font 7 loaded
Font 8 loaded
Smooth font enabled
Display SPI frequency = 2.50
Touch SPI frequency = 2.50
[/code]
```

Bodmer, 2022-01-24:
The problem with the C3 variant is likely to do with the SPI configuration. I do not have an ESP32 C3 so am unable to test any changes. At the moment the C3 is no supported. If you can figure out what the problem is then do let me know.

gabrielcsapo, 2022-01-26:
FSPI=0
HSPI=1

Looking at debugging this further tonight.

gabrielcsapo, 2022-01-28:
If anyone has a c3 and wants to help figure this out https://github.com/Bodmer/TFT_eSPI/pull/1586 I have a draft PR opened to collaborate.

Bodmer, 2022-01-28:
The library uses register level access to the SPI registers. Looking at the ESP32 board package I see the C3 requires extra code before triggering the transmit. At the moment the library does not have this, so that is likely to be the problem:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c#L734-L737

gabrielcsapo, 2022-01-28:
Thanks @Bodmer , I will look into this.

gabrielcsapo, 2022-01-28:
@Bodmer I have added this https://github.com/Bodmer/TFT_eSPI/pull/1586/commits/92668553f4d8082e5da38af7aea170c9b6227d79 which seems to do exactly what the lines you sent do. Still no dice.