Bodmer/TFT_eSPI #1431: Touchscreen MOSI seems hardcoded (ESP32)

daxliniere, 2021-11-16:
I have just finished chasing my tail, trying to get the touch screen of an ILI9488 working on ESP32 (DevKitC v4 38-pin).
I had chosen non-standard SPI pins and customised them in **User_Setup.h**. The display was working perfectly, but when I tried any touchscreen sketches, the calibration phase would race through without being touched and obviously the touch didn't work once the main sketch loaded.

For other reasons, I ended up switching to standard SPI pins (18, 19, 23) and miraculously, the touch stuff all works now.

Is it possible that the SPI pins for the touch screen are hard-coded to the defaults rather than referencing the pins specified for the TFT?

All the best,
Dax Liniere.

Bodmer, 2021-11-17:
The touch SPI code uses the same SPI pins as the display for the interface so it is not hardcoded to a default port.

As [reported here ](https://github.com/Bodmer/TFT_eSPI/discussions/898) the ILI9488 SPI output ( SDO/MISO) does not go tristate and this cannot be circumvented, so do not connect the SDO/MISO pins on an ILI9488 if other SPI devices are on the same SPI bus.

You can use the "Test_Touch_Controller" diagnostic sketch to see if the touch screen can be read.

daxliniere, 2021-11-17:
Hey @Bodmer, thanks for the reply. Yes, I had already read about that and your clever fix to the diode, but in my case I never had ILI9488's SPI output connected to anything.

Any thoughts now that you know this?

Bodmer, 2021-11-18:
Have you connected it exactly as shown here:
https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/RPi_TFT_Connections.png

Bodmer, 2021-11-18:
Run the Test_Touch_Controller sketch to see if communications are working with the touch controller when alternative pins have been allocated.

Run Read_User_Setup to check pin allocations are being correctly picked up by the compiler.  List the pins you were trying to use.

Bodmer, 2021-11-27:
Assumed resolved.

closed 2021-11-27 by Bodmer

fungreenfox, 2022-01-19:
I am having 3x ili9488 at 3.5" with touch. Only one of these are working in touch using standard setup wiring. Only change of rst to pin 4, but it doesnt matter how i set the wiring, even moving it all to pin 12, 13 and 14 for miso and miso and clk works.
There is a fault in the display and i ordered 2 new, each from different seller. Only difference from the first one, are some banes at the backside of the display are going downwards, instead of upwards. The 2 displays are having touch_cs shortcutted, and i have tried any combination with no luck.

Bodmer, 2022-01-19:
The link has a bad description, this suggests it is a SPI display, but some pictures show parallel boards. It is not clear to me which display boards you have.

The touch function requires a chip select for the TFT and one for the touch controller.

fungreenfox, 2022-01-19:
I have this one, as it says: 3.5 inch TFT LCD Module with Touch Panel ILI9488 Driver 320x480 SPI port serial interface (9 IO) Touch ic XPT2046 for ard stm32

Can you specify chip select for TFT and one for the touch controller? is this a chip needed to be added or a wiring?

Bodmer, 2022-01-19:
The chip select pins for the TFT and touch controller MUST be different. Do not connect TFT MISO as it does not go tristate on the ILI9488.  See here:
https://github.com/Bodmer/TFT_eSPI/discussions/898

fungreenfox, 2022-01-20:
I have readen that #898 post with pleasure, but look at this; there is no D1 diode, and the display has the same problem.
https://www.dropbox.com/s/apfze1hanbz83lq/ili9488V1.jpg?dl=0

Sampozzo, 2022-04-09:
Hello, i have the same issue, i have two ILI9488 marked v1.0 **without** the diode described there: https://github.com/Bodmer/TFT_eSPI/discussions/898

I can't get them working, with or without MISO connected. The calibration test example will autopress the corners and give the result:

```
// Use this calibration code in setup():
  uint16_t calData[5] = { 1, 1, 1, 1, 0 };
  tft.setTouch(calData);
```

Any help on this? @fungreenfox have you solved?

fungreenfox, 2022-04-10:
Sampozzo, I have returned the issue to the seller, and he is happy to return 20% of my cost, if I send the display back. So thats a no go. I bought two more (9488 and 5"), and they are fortunately version 2, and they are working correct, even if placing the touch_cs on another pin. Pins used are:
Do only define pins in user_setup.h:  
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC   2  // Data Command control pin
#define TFT_RST  4  // Reset pin (could connect to Arduino RESET pin)
#define TFT_BL   25  // LED back-light (required for M5Stack)
#define TOUCH_CS 5   // Chip select pin (T_CS) of touch screen

User_Setup_Select.h:
#ifndef USER_SETUP_LOADED
#include <User_Setup.h>  

In Setup21_ILI9488.h
only #define TFT_BACKLIGHT_ON HIGH  // Level to turn ON back-light (HIGH or LOW) and no other pins repeated. This of my suggestion can go down, if pins are defined in more places like setup and the Setup21_ILI9488.h itself. 
Actually this runs, with a calibration that saves the result to SPIFF, and does not need to be commented in setup, but i got this result that works for a 5" 9488 (can vary from display to another)
uint16_t calData[5] = { 417, 3503, 341, 3516, 3 }; // and should not be just 1's.

The display worked for a weeks with no trouble, though the breadboard project was paused for a month, and turning back , the display did this act of shortcutting the touch_cs. I ripped all wires off the board and placed them again. Re-calibrated the savefile, and made sure, that there were no incorrect pins in use. (try disconnect all but display until it works).

But again, version 1 was a no-go for me. I asked 4-5 sellers, if they have the version 2, and nobody even knows of it. But the two ordered displays are version 2. 
And the miso pin? yes, its connected with no conflicts.

Sampozzo, 2022-04-10:
Hi @fungreenfox, thank you very much for your detailed reply, i will try your code, but @Bodmer said it is not pin related but hardware. So no solution...

Bodmer, 2022-04-10:
@Sampozzo provide a link to each of the displays you have and list all connections between the ESP32 and each display type. I am not sure where I said it is pin related!

Sampozzo, 2022-04-10:
@Bodmer you said it is **NOT** pin related but hardware, in your first reply on this post  :)

Bodmer, 2022-04-10:
Must be a misunderstanding there!

Here are example pin connections. Note that the display SDO MUST be left disconnected.

[image]

Sampozzo, 2022-04-10:
YESSSSSSSSSSSS!!!!!!
I left unconnected T_DO and SDO, looking at your picture now, i connected T_DO and it works!!!!!!

Thank you very much @Bodmer

Olivierdemoustier, 2023-02-13:
I have bought a very nice Development Board with touch screen and ESP32 , all in one. 
But SPI for display use different pins as SPI for touch. How can I configure this in User_Setup.h ?

MichaelSaan, 2023-05-18:
I have the following question about this, GPIO21 is also required for the I2C bus (SDA). After connecting the display, can I still use the I2C bus for additional I2C slaves on the ESP32?

Bodmer, 2023-05-19:
The GPIO pins connected to the display cannot be used for I2C as well. Use a different GPIO for TOUCH_CS and update the setup file accordingly.