Bodmer/TFT_eSPI #1434: Corrupted graphics troubleshooting

john-NZ, 2021-11-17:
Hi All

First of all a big thank you to Bodmer for the library.

I am using ESP32 dev board and 480x320 Rpi LCD TFT  with driver defined #define RPI_ILI9486_DRIVER 

I am seeing very corrupted graphics on all of the 480x320 examples and I can't solve the issue.  

I took a brief video of the familiar graphics test.
https://photos.app.goo.gl/BQ2Z2sZzzKQoQ48j7

There seems to be an issue with the resolution or rotation ?   But I can't solve it.
  Serial.println(tft.height());   prints 480 to serial port
  Serial.println(tft.width());    prints 320 to serial port

Has anyone seen this before?

Many Thanks

John

Gabryxx, 2021-11-17:
// Customised User_Setup files are stored in the "User_Setups" folder.

#ifndef USER_SETUP_LOADED //  Lets PlatformIO users define settings in
                          //  platformio.ini, see notes in "Tools" folder.

// Only ONE line below should be uncommented.  Add extra lines and files as needed.

//#include <User_Setup.h>           // Default setup is root library folder

//#include <User_Setups/Setup1_ILI9341.h>  // Setup file configured for my ILI9341
//#include <User_Setups/Setup2_ST7735.h>   // Setup file configured for my ST7735
//#include <User_Setups/Setup3_ILI9163.h>  // Setup file configured for my ILI9163
//#include <User_Setups/Setup4_S6D02A1.h>  // Setup file configured for my S6D02A1
//#include <User_Setups/Setup5_RPi_ILI9486.h>        // Setup file configured for my stock RPi TFT

Only 1 need uncommented so...
//#ifndef USER_SETUP_LOADED //  Lets PlatformIO users define settings
#include <User_Setups/Setup5_RPi_ILI9486.h>        // Setup file configured for my stock RPi TFT

is correct?
i use all examples with Esp32 Dev Kit and Ili 9488 touch without problem...the only problem is with Screenshot save routine that give a black screen on Processing under Win 10

Bodmer, 2021-11-17:
Hi John, it looks like the display is not being initialised correctly. This may be due to a number of reasons. To narrow down the reasons please post a link to the display sellers product page.

john-NZ, 2021-11-18:
Hello Bodmer

Thank you for the help, here is a link to the manual
https://www.jaycar.com.au/medias/sys_master/images/images/9592783667230/XC4631-manualMain.pdf

There is a repo for this LCD

I have tried to confirm which driver is used but I can't confirm it.    There is a repo for this LCD, but still I can't quite understand if the driver is: ILI9486
https://github.com/Jaycar-Electronics/XC4631

Bodmer, 2021-11-18:
It is not clear which display controller the board has. Have you connected the display exactly as shown here:

https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/RPi_TFT_Connections.png

john-NZ, 2021-11-19:
[User_Setup.txt](https://github.com/Bodmer/TFT_eSPI/files/7569164/User_Setup.txt)

Hello Bodmer, I checked the connections against your photo, they are exactly the same.  

My setup file is attached.

I am fairly sure the board is from Waveshare because the git repo mentions waveshare here:
https://github.com/Jaycar-Electronics/XC4631/commit/c78671d85afdd7422735d13c321e28a040d7911a

I made a very simple program:  The result is here:
https://photos.app.goo.gl/muM1h1ggiRC4Yupo9
The code is below

I have tried the following drivers:
HX8357D
ILI9341
ST7796
RM68140
R61581

Is there anyway to discover the driver?

#include <SPI.h>
#include <TFT_eSPI.h>       // Hardware-specific library
TFT_eSPI tft = TFT_eSPI();  // Invoke custom library

void setup(void) {
  tft.init();
 }

void loop() {  
  tft.fillScreen(TFT_GREEN);
  delay(5000);
  tft.fillScreen(TFT_BLUE);
  delay(5000);
    tft.fillScreen(TFT_RED);
  delay(5000);  
}

john-NZ, 2021-11-19:
Hi Bodmer

Quick update....   I was able to fix the corrupt graphics by lowering the SPI frequency.  I think the ILI9486 max frequency is 32Mhz. 

I had to reduce the frequency of my board to 20MHz.  Now the speed is a little too slow for my needs, but no corruption.

I assume the limiting factor is the driver chip, rather than the ESP32 software SPI implementation?    

Trying to implement hardware  SPI would not help in this case would you agree?

Thanks for helping me, I am glad to finally have a working LCD!  I will try to work with the slower speed.

John

Bodmer, 2021-11-21:
20Mhz is the limit for that RPi display. It is a limitation of the boards circuit design rather than the library or display.

Later RPi boards have a better single chip interface design and work at up to 80MHz.

Glad you have it working.

Incidentally the board you have does not support reading from the display.

closed 2021-11-21 by Bodmer