Bodmer/TFT_eSPI #1436: I can't show .jepg image in sd card at tft screen.

EakUnicycle, 2021-11-18:
I'm using esp32 and tft 3.5".
I can use the example TFT_flash_jpg.ino.
But I can't use the example ESP32_SDcard_jpeg.ino.
My computer screen shows the data of the image file in the sd card.
But my tft display is not showing pictures.
What is the problem and how do I fix it ?

[image]

[image]

Bodmer, 2021-11-18:
There are various SD libraries for the ESP32 so it is best to test SD card access first using the SD library examples before trying to use the TFT_eSPI example.  Make sure that the SD card library example can read the SD card files.

Note also that the SD card pins on the TFT are at the other end of the PCB and need to be wired up if you are using the card holder on the display.  Some displays have series resistors on the SD card data lines and these may mean that the SPI clock rate for the SD card will have to be reduced (see SD library documentation as I cannot advise here.)

Lastly you may need to call tft.endWrite() before calling any SD card functions to make sure SPI transfers to the TFT have been completed.

I have found that power cycling the ESP32 corrupts the SD card sometimes and makes it unreadable. Maybe a pullup on the SD card chip select would help here, however pulling the power plug when an SD access is under way could also corrupt the card.

EakUnicycle, 2021-11-19:
Thank you for giving advice.
I found that you updated the library. 
I switched to the updated library from you. 
I can use the program ESP32_SDcard_jpeg.ino has succeeded.

closed 2021-11-19 by Bodmer