Bodmer/TFT_eSPI #1437: Library "forgets" rotation value

MakerMatrix, 2021-11-18:
I have ported a code over from Bodmer/TFT_ILI9341 running on Atmel (Leonardo, specifically), to this library, running on RP2040.

Most of the port went easily.  However I have noticed in the ported version, the application "forgets" something about the pixel coordinate dimensions.  I start by setting tft.setRotation(3) (or 1) in order to be dealing in a landscape coordinate frame.  I'm able to for example render text across the entire 320 pixel "width" of the screen.  Then, for some currently unknown reason, the library starts rendering in that dimension as if the screen were still rotated landscape mode, but only had 240 pixels in the long dimension.

Currently I'm working around this by putting additional tft.setRotation() calls in the code, but would really like to find out what causes it to revert to the wrong resolution and fix that instead.  Any idea what calls I'm using might affect this, or even if there are calls I can use to help debug?  At first I though tft.fillScreen() was triggering it but I think I've ruled that out.

Bodmer, 2021-11-18:
This should not happen.  Do the TFT_eSPI standard examples run OK?

Try to simplify the sketch to narrow down where the problem occurs.

Post an example Arduino IDE compatible sketch that demonstrates the problem and I can test (note I do not have an hx711 load cell board.

Bodmer, 2021-11-26:
I had a look at the TFT_eSPI_Charts code and see that the "begin" functions sets the screen rotation and grabs a copy of the TFT width and height.  See here:
https://github.com/MakerMatrix/TFT_eSPI_Charts/blob/master/src/TFT_eSPI_Charts.cpp#L16-L18

 If you then change the rotation in your own code... the chart library will still be using the old width and height values.

I therefore suspect this is not a problem with the TFT_eSPI library.

closed 2021-11-26 by Bodmer