Bodmer/TFT_eSPI #1441: Possible issue with last version 2.3.80 on TTGO T-Display

hpsaturn, 2021-11-19:
## Description

I did return back to version 2.3.73, because the latest version (2.3.80) leave the screen in black.

## PlatformIO Config

```python
[tft_common]
extends = esp32_common
lib_deps = 
	${common.lib_deps}
	bodmer/TFT_eSPI @ 2.3.73

[env:TTGO_TDISPLAY]
extends = tft_common
board = esp32dev
lib_ldf_mode = deep
lib_ignore = gui-utils-oled
build_flags = 
	${common.build_flags}
	-Os
	-DUSER_SETUP_LOADED=1
	-DST7789_DRIVER=1
	-DENABLE_TFT=1
	-DTFT_WIDTH=135
	-DTFT_HEIGHT=240
	-DCGRAM_OFFSET=1
	-DTFT_MOSI=19
	-DTFT_SCLK=18
	-DTFT_CS=5
	-DTFT_DC=16
	-DTFT_RST=23
	-DTFT_BL=4
	-DTFT_BACKLIGHT_ON=1
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=40000000
	-DSPI_READ_FREQUENCY=6000000
```

zingo, 2021-11-19:
Had the same problem the new 2.3.81 version seem to work.

Bodmer, 2021-11-21:
2.3.80 had a bug, please use 2.3.81

closed 2021-11-21 by Bodmer