Bodmer/TFT_eSPI #1444: Pico ResTouch-LCD-2.8 with ESP32 WROVER ( black screen )

misiek303, 2021-11-22:
Pi Pico is very limited and wanted to use ESP32 anyway. But I know the Display works.
https://www.waveshare.com/pico-restouch-lcd-2.8.htm

I switched to ESP32 WROVER but I got black screen. This screen supposed to work with ST7789.

**my User_setup.h**

#define ST7789_DRIVER
**#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   21
#define TFT_DC    2
#define TFT_RST   4**
#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT
#define SPI_FREQUENCY 8000000L
#define SPI_READ_FREQUENCY  2000000
#define SPI_TOUCH_FREQUENCY  2500000

1. I tried most of the drivers and many different frequencies, but screen does not come to life.
2. I am confused about CLK and SDIO_CLK, I tried them both but not sure which one I should actually connect to ESP PIN 18.
3. I don't have LCD_BL connected to anywhere.

Any ideas ?

misiek303, 2021-11-23:
I made some progress. I have digital clock running, **but the the screen is very very dark, you can barely see the clock.**

#define ST7789_DRIVER
#define TFT_WIDTH  240
#define TFT_HEIGHT 320
#define TFT_BACKLIGHT_ON HIGH
#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   14 
#define TFT_DC   27
#define TFT_RST  33 
#define TFT_BL   32
#define LOAD_GLCD
#define LOAD_FONT2
#define LOAD_FONT4 
#define LOAD_FONT6
#define LOAD_FONT7
#define LOAD_FONT8
#define LOAD_GFXFF
#define SMOOTH_FONT
#define SPI_FREQUENCY 8000000L
#define SPI_READ_FREQUENCY  6000000
#define SPI_TOUCH_FREQUENCY  2500000

Turning #define TFT_BACKLIGHT_ON to LOW, make screen totally black.

Any ideas ?

misiek303, 2021-11-23:
It needed additional 5V. I guess 3.3v for logic, 5V for BL.

misiek303, 2021-11-23:
And this to display colors properly
define TFT_RGB_ORDER TFT_BGR

Thanks. Nice chatting :)

Bodmer, 2021-11-24:
Glad you found the solution! Apart from the pins the other setup settings should be the same as the RPi Pico.

closed 2021-11-24 by Bodmer