Bodmer/TFT_eSPI #1447: display a jpeg with transparency

infrafast, 2021-11-24:
Hello, I am using the jpeg decoder library (https://github.com/Bodmer/JPEGDecoder) together with the TFT-eSPI 

This works very ok, but I was wondering how I could apply transparency to the rendering of the image

I guess I should use AlphaBlending function in the bool tft_output(int16_t x, int16_t y, uint16_t w, uint16_t h, uint16_t* bitmap)
but since this function render a block and not pixel by pixel I couldn't find a solution...

Maybe this can be achived easily, would you have any example to provide? Thanks a lot. 

ps: My final requirement is to be able to fade in / fade out the image

Bodmer, 2021-11-24:
Hi, I have not have an example. The pixel block consists of  w*h aray pixels in memory, each pixel is a 16 bit colour so you could copy each pixel to a new memory block and use the alphablend function.

Bodmer, 2021-11-24:
Try this
[Flash_Jpg_Fade.zip](https://github.com/Bodmer/TFT_eSPI/files/7598637/Flash_Jpg_Fade.zip)
:

closed 2021-11-24 by Bodmer

infrafast, 2021-11-25:
great! works perfectly thank you!