Bodmer/TFT_eSPI #1448: Issues with big sprite rotation speed...

exdiogene, 2021-11-24:
Hi,

 first I want to thank you very much for the excellent library you provided!

I have used your library for several months now and having some issues with pushRotated() and pushImage() at 90° or 270°.

I am using a ESP32-WROVER-8 CPU on a WT32-SC01 V3.2 display board.

I am trying to rotate a sprite 256 X 256 16 bits containing a decoded PNG image.

The sprite was created by your library in the 4MB of lower PSRAM.

If  I use a normal pushImage() of that sprite it take approximately 65 ms to display.

But if I try to pushRotated(90) or  pushRotated(270) it will take 248ms.

Surprised by this long process, then I tried to decode each of the PNG 256 pixels lines rotated in the sprite with 256 vertical pushImage() and after I pushImage() on the screen it has taken 337ms.

I need to achieve a minimum 6fps speed at 0°, 90°, 180° and 270°, so it should take less than 166mS by frame.

Then I figure out that the issue seem related to the use of pushImage() for a vertical line vs an horizontal line!

It also seem to take a lot more time with the SPI overhead for a vertical line.

Is there some solution to accelerate de process of copying vertical lines to a sprite?

Thank you for your time,

 Rej

Bodmer, 2021-11-25:
Yes, pushRotated will be slower as it uses a coordinate transform using Sin, Cos and fixed point maths so a lot of calculations are involved for an arbitrary rotation angle. Rotations of 90 deg. icrements could clearly be done without the maths overhead.

This is not a generalised bit of code but it will rotate a sprite by 90 quite quickly; for 180 and 270 you can adapt:

```
uint16_t* sprPtr = nullptr; // Global

...
  sprPtr = (uint16_t*)spr.createSprite(WIDTH, HEIGHT); // Pointer to sprite memory start

...
      // Create a line buffer and copy pixels in sprite
        uint16_t lineBuf[spr.height()];
        tft.startWrite();
        for (uint16_t px = 0; px < spr.width(); px++) {
          for (uint16_t py = 0; py < spr.height(); py++) {
            lineBuf[spr.height() - py - 1] = sprPtr[px + py * spr.width()];
          }
         // Use pushImage as it will crop off screen areas correctly
          tft.pushImage(x, y + px, spr.height(), 1, (uint16_t*)lineBuf);
        }
        tft.endWrite();
```

closed 2021-11-25 by Bodmer

exdiogene, 2021-11-25:
Yes I know that pushRotated() is supposed to be slower with the pixels rotation, so this is why I used pushImage(). But pushImage() is more than 5 times slower for a vertical line of pixels compare to an horizontal line !

Here is a part of my sprite filling code, this code is executing in 65mS when image_rot is 0, but 337mS when image_rot is 90 :

`for (y = 0; y < spr.height(); y++) {
    temp_millis = millis();
    png.getLineAsRGB565(pDraw, bpm_data2, PNG_RGB565_BIG_ENDIAN, 0xffffffff); //Decode an horizontal line of 256 pixels
    decode_time += millis() - temp_millis;
    temp_millis = millis();
    if(image_rot == 0){
      screen.pushImage(0, pDraw->y+i, pDraw->iWidth, 1, bpm_data2); //Push one line horizontally 
    }else if(image_rot == 90){
      reverse_buf(bpm_data2, pDraw->iWidth);
      screen.pushImage(256- pDraw->y+i, 0, 1, pDraw->iWidth, bpm_data2); //Push one line vertically 
    }else if(image_rot == 180){
      reverse_buf(bpm_data2, pDraw->iWidth);
      screen.pushImage(0,256 - pDraw->y+i, pDraw->iWidth, 1, bpm_data2);
    }else if(image_rot == 270){
      screen.pushImage(pDraw->y+i, 0, 1, pDraw->iWidth, bpm_data2);
    }
    display_time += millis() - temp_millis;
}
`
**screen** is the **tft** 256 X 256  sprite I use to compose the image before copying it to the display.

The reverse_buf() function is used to reverse the order of pixels in the buffer to keep the text correctly displayed in the sprite and use a total of only 4mS for all the 256 lines. (0.016 mS per line)

As you can see, the only difference in the code is that pushImage() send a vertical line when image_rot is 90...

So why is it taking more than 5 times to execute ?

Bodmer, 2021-11-25:
There is no reason that pushImage speed will change significantly. Try to narrow down where the delay is introduced. Maybe this is down to cache or PSRAM buffer flushing caused by the reverse_buf function?

Bodmer, 2021-11-25:
Just tried this simple bit of code:
```
          tft.startWrite();
          uint32_t dt = micros();
          for (int i = 0; i < 1000; i++) {
            tft.pushImage(0, 0, 256, 1, (uint16_t*)lineBuf);
          }
          Serial.print("H ="); Serial.println(micros()-dt);
          yield();
          dt = micros();
          for (int i = 0; i < 1000; i++) {
            tft.pushImage(0, 0, 1, 256, (uint16_t*)lineBuf);
          }
          Serial.print("V ="); Serial.println(micros()-dt);
          yield();
          tft.endWrite();
```
Results are:
```
H =108932
V =108922
```
So almost identical at 108.9 us per line (40MHz hardware SPI clock, 37.6Mbps actual rate) as I would expect. This is with a 16 bit TFT display.

I am suspicious of the "(0.016 mS per line)" value quoted, which would imply only 16 us per line and require a 256MHz clock for 16 bit colours.

What hardware are you using and which display driver?

Bodmer, 2021-11-25:
I just noticed that "screen." is actually a sprite in PSRAM. I therefore suspect this is a PSRAM cache flush problem and not the sprite pushImage function.

The problem is that PSRAM is serial accessed, and is read and buffered by the processor, so you will get a large number of cache flushes if vertical lines are written. Assuming the PSRAM cache is say 32kbytes then it needs to be filled 4 times for incrementing addresses (256 horizontal lines), but 256 x 4 = 1024 times for 256 vertical lines.

On a line by line basis this means 1 cache fill for a single 256 pixel horizontal line and 4 cache fills for a 256 pixel vertical line.

You would be better off not rotating in the sprite, but only rotating when pushing the sprite to the screen as suggested by the sample code above.

exdiogene, 2021-11-25:
I have already tried to delete the reverse_buf function and it did not have any significant effect on the delay else than a 4mS reduction...

Here is the code : 

> void reverse_buf(uint16_t *data, size_t n)
> {
>     size_t i;
>     for (i=0; i < n/2; ++i) {
>         int tmp = data[i];
>         data[i] = data[n - 1 - i];
>         data[n - 1 - i] = tmp;
>     }
> }
> 

Like I previously stated, I use a WT32-SC01 V3.2 display board, using a ESP32-WROVER-8 CPU.

CPU Frequency 240MHz.
Flash Frequency 80MHz.
Flash Mode QIO.
PSRAM enabled.

This is the User_Setup_Select.h selected line for your library : 

> #include <User_Setups/Setup201_WT32_SC01.h>        // Setup file for ESP32 based WT32_SC01 from Seeed

Their display is connected through a SPI interface.

I do not know anything about the cache on these module, maybe you can shed some light on this topic...

By the way, how can I include CODE like you do here, each time I try each NewLine are lost and the code look packed together?

This is the reason why I use the QUOTE instead, to keep the NewLines.

Bodmer, 2021-11-25:
To add code you put 3 backticks - then the code and then 3 backticks, Usually the backtick is on the key next to the number 1.  PSRAM caching is explained in my last post above. This is not well documentated but is likely to be the root cause.

Bodmer, 2021-11-25:
Format syntax info here:

https://docs.github.com/en/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax

exdiogene, 2021-11-25:
I just read your last reply, I was suspecting that it could be related to SPI access delays.

The problem I have is that I must copy the PNG image into the sprite and after that include a lot of drawings on it before pasting it on the screen. The screen is already accessed with SPI, so I do not have many choices for the operations...

exdiogene, 2021-11-25:
I really do not understand how it is possible that the vertical lines, even with 4 times more accesses with SPI, could take 5 times the time compare to horizontal lines.

Do you know a trick to alleviate the burden of that delay?

exdiogene, 2021-11-26:
When I click "<>" "Insert Code", it only insert 2 BackTicks. This must be the issue.

I just notice that when I select the code in the form and click afterward it include 3 starting and 3 closing BackTicks.

Thank you for the information.

exdiogene, 2021-11-26:
To give you a clear idea of my project, I copy a Map Tile to a 256 X 256 PSRAM sprite, then I add some information on the Map Tile including a vehicle and a compass. Only 4 angles are permitted for the sprite background 0, 90, 180, 270.

I am actually limited with the slow process of SPI interface for the PSRAM sprite and the display.

If I can resolve the slow process of pushImage() of a vertical line, I would be OK.

If it took only twice the time of an horizontal line, it would be acceptable.

Bodmer, 2021-11-26:
As an experiment reduce the sprite size to less than the PSRAM cache size and try it. I think a 200 x 200 sprite in PSRAM will not hit the cache delays.

Ways around the hardware limitations are either to split the sprite into 4 smaller parts, or have 4 PNG images, one with each rotation you need, or 4 master sprites with the image already rotated (you have plenty of PSRAM for this.

Bodmer, 2021-11-26:
Another probably better approach is to create a rotated vertical strip image buffer in normal RAM that is more than 1 pixel wide, say 16 pixels wide, and then use pushImage with a width of 8 instead of 1. This will run much faster.

exdiogene, 2021-11-26:
Thank you for your solutions, but it seem that they cannot be applied to my project.

Unfortunately I have to use 256 X 256 PNG tiles because these are the only available on the market. There are 32GB of tiles on my SD card for my GPS project and it take an average of only 16mS to load one complete image.

The PNG decoder library send one horizontal decoded line to the display interface function and I must abide by those rules.

I already have a one line buffer in normal RAM, But sometime the displacement of the vehicle on the map would use 4 segmented images on the display, so the size horizontally or vertically of each segment of image can change.

Anyway, at the end I must copy all those segments over the sprite 256 X 256 that your library created in PSRAM. This way I can copy the sprite with all the added details on the display with less flicker!

The burden that I have is that the WT32-SC01 board is using SPI for its display, the CPU use SPI for its PSRAM and I must have another SPI for the SD card reader. All these SPI create delays that I could manage until de huge delay for vertical lines...

If I could manage the PSRAM cache by myself it could be a solution, but I was unable to find any information about the way it is operating on the WEB...

I have noticed that you use "tft.startWrite()" and "tft.endWrite()" to enclose the multiple "tft.pushImage()" operations. I was not doing that with my sprite filling operations, I just tried to enclose the 256 sprite "screen.pushImage()" with "screen.startWrite()" and "screen.endWrite()" and it resulted with a BLANK display. I am not familiar with this kind of coding, can it reduce the process overhead?

Thank you very much for your help,

 Rejean

exdiogene, 2021-11-26:
I have made tests with different size of the "screen" sprite, without modifying the rest of the code, and discovered the culprit : 

```
128 X 128 : horizontal lines = 17 mS, vertical lines = 25 mS
200 X 200 : horizontal lines = 41 mS, vertical lines = 63 mS
240 X 240 : horizontal lines = 58 mS, vertical lines = 73 mS
248 X 248 : horizontal lines = 63 mS, vertical lines = 116 mS
256 X 256 : horizontal lines = 69 mS, vertical lines = 335 mS

```
For the last 16 pixels of the vertical lines, it take over 5 times to copy the lines.

Now I have to take a big decision : 
1 - I modify my 4000 code lines to allow only a 240 X 240 pixels copy of 256 X 256 pixels Map Tile image.
2 - I try to find the reason of the behavior for the last 16 pixels and fix it.
3 - I try to find out new hardware without those limitations.

Because I already have invested over 500 hours of work with the actual hardware and code, the easiest solution would be option 2 if I can find the portion of the ESP32 code causing this behavior.

If you  have any idea where to search the problem, I would greatly appreciate any suggestion!

Thank you in advance,

 Rejean

P.S. You should mention this behavior with your library, in case someone else try to rotate a sprite over 240 X 240.

Bodmer, 2021-11-27:
To solve this in the board package you will need to delve down deep into the IDF code and see how it buffers data for the PSRAM. I suspect if you follow this route you will run into further difficulties as you will need a detailed understanding of how much normal RAM is available.

I suspect you have not understood my suggested for a larger vertical buffer, a 1 pixel vertical line buffer is inefficient as it will need to be written 256 times to the PSRAM sprite! My suggestion is to use a wider buffer, eg. This would be a very simple change to your code and should give the performance boost you seek.

exdiogene, 2021-11-27:
Yes I understood your suggestion for the larger vertical buffer, but in my project it would be unfeasible.

The image in the sprite is composed of horizontal lines that become available one by one from individual function calls of the PNG file decoder.

At first it is one whole image that is available, but when the vehicle go 1 pixel North-East, the sprite would be composed of 4 images decoded one by one to fill the sprite.

There will be 1 bottom line of 255 pixels of the top image, under it 255 top lines of the bottom image and at the right there will be 1 left pixel of the top right image and one vertical line of 255 top pixels of the right bottom image.

After a while it will look like this : 

```
_______________
|___ top___|-r-|
|----------|-i-|
|--bottom--|-g-|
|----------|-h-|
|__________|_t_|

```
So there would be different image proportions displayed, I hope you can understand the problem with your solution.

There would be 0 to 255 vertical lines to copy for each of the 4 images used, but they will never be together and I do not know their count in advance because the GPS could jump to a difference of 1 to 16 pixels depending on the vehicle speed...

The drawing section is like this  : 
```
void PNGDraw(PNGDRAW *pDraw) {
minmax my_minmax;
  if(pDraw->pUser){
    my_minmax = *((minmax*)pDraw->pUser);
    if(my_minmax.ymin <= pDraw->y && my_minmax.ymax >= pDraw->y){
      png.getLineAsRGB565(pDraw, bpm_data2, PNG_RGB565_BIG_ENDIAN, 0xffffffff); //Decode an horizontal line of 256 pixels
      if(my_minmax.offset < 0){
        if(my_minmax.edge){
          if(image_rot == 0){
            screen.pushImage(256+my_minmax.offset, pDraw->y+i, -my_minmax.offset, 1, bpm_data2); //Push one line horizontally
          }else if(image_rot == 90){
            reverse_buf(bpm_data2, -my_minmax.offset);
            screen.pushImage(pDraw->y+i, 256+my_minmax.offset, 1, -my_minmax.offset, bpm_data2); //Push one line vertically
          }else if(image_rot == 180){
            reverse_buf(bpm_data2, -my_minmax.offset);
            screen.pushImage(0,256 - pDraw->y+i, -my_minmax.offset, 1, bpm_data2);
          }else if(image_rot == 270){
            screen.pushImage(256- pDraw->y+i, 256+my_minmax.offset, 1, -my_minmax.offset, bpm_data2);
          }
...
}
```

So I would need to search about how they buffers data for the PSRAM... :(

exdiogene, 2021-11-30:
I tried to search for information about PSRAM or SPI data buffer usage and found nothing on the Web...

Seem that Espressif are not giving much information about these in deep details!

I have started to modify my code but it is a very long process, with many difficulties... :(

It would be so helpful if Espressif was giving the details about that buffer, so we could increase it just a little bit for usage with images bigger than 240 X 240. Most of the current display are 320 X 240 or bigger, so it is a must!

If anyone have any information, I would greatly appreciate any feedback, but it seem that the "**issue**" have been closed, so I believe that no one else than you Bodmer could see my messages... :(