Bodmer/TFT_eSPI #1449: Custom Eye on the Animated Eye example.

BishoPower, 2021-11-26:
Can anyone give me some pointers on going about creating a custom eye for the animated eye example?

Bodmer, 2021-11-27:
You need to create a set of image files or adapt the ones provided. Example images are here:
https://github.com/adafruit/Uncanny_Eyes/tree/master/convert/defaultEye

The python script to convert these images to a header file for inclusion in the sketch are here.
https://github.com/adafruit/Uncanny_Eyes/tree/master/convert

The sketch is adapted from on by Adafruit, so you may find help on the Adafruit website.

Bodmer, 2021-11-27:
Have a look here also as someone has created a custom eye and describes the process:
https://makezine.com/projects/eye-of-newt-keep-watch-with-a-creepy-compact-animated-eyeball/

closed 2021-11-27 by Bodmer