Bodmer/TFT_eSPI #1455: ILI9488 esp32

HrvojeSusilovic, 2021-11-27:
When I upload example in Test_Touch_Controller and open serial monitor in Arduino Ide, numbers listed below start coming up, as if I was holding the screen with my finger
....
x: 66     y: 2788     z: 3329 
x: 83     y: 2854     z: 3259 
x: 88     y: 2811     z: 3379 
x: 77     y: 2762     z: 3300 
x: 82     y: 2752     z: 3277 
x: 85     y: 2709     z: 3278 
x: 85     y: 2748     z: 3323 
x: 85     y: 2802     z: 3378 
x: 41     y: 2799     z: 3403 
x: 87     y: 2803     z: 3371 
x: 85     y: 2837     z: 3249 
x: 164     y: 2776     z: 3348 
x: 77     y: 2783     z: 3375 
x: 84     y: 2747     z: 3327 
x: 85     y: 2764     z: 3295 
....

Bodmer, 2021-11-27:
The ILI9488 SPI output ( SDO/MISO) does not go tristate so do not connect the TFT screen interface SDO/MISO pin on an ILI9488 if other SPI devices are on the same SPI bus.

HrvojeSusilovic, 2021-11-27:
Hmm I have two same display with one work fine with other dont work.. ( I think dont work correct. How I am explain in first post.) 
I am contact seller and he say me: "(Well, for one of the screens I had to use the code inside the folder 3.5inch_SPI_Module_ILI9486_MSP3520_V1.0 that you can download from lcdwiki http://www.lcdwiki.com/zh/3.5inch_SPI_Module_ILI9488_SKU:MSP3520 and for the other two screens, I had to use the code inside 3.5inch_SPI_Module_ILI9488_MSP3520_V1.1. There are several differences in the lcd_init function (and the LCD_direction one) inside the file lcd.c" )"

Bodmer, 2021-11-27:
It appears you have one ILI9488 display and one ILI9486 display. These need different TFT drivers.

On the ILI9488 do not connect to the SDO pin on the display.

The ILI9488 and touch functions work fine together as demonstrated by this user:
https://github.com/DustinWatts/ESP32_TFT_Combiner

HrvojeSusilovic, 2021-11-28:
I work this touch deck and i have this PCB board... and how i am say one work fine other dont. then i try cut SDO.

HrvojeSusilovic, 2021-11-28:
now I see on this pcb SDO is  n.c.

Bodmer, 2021-11-28:
Run the Read_User_Setup sketch to make sure the compiler is picking up the right settings.

It is not clear to me what is not working. Is it only touch that is not working? Or is it the screen?

Post web links to the vendors product page.
The ILI9486 requires a slower TFT clock rate than the Ili9488. Try 10MHz.

HrvojeSusilovic, 2021-11-28:
Run the Read_User_Setup sketch to make sure the compiler is picking up the right settings.
[image]

It is not clear to me what is not working. Is it only touch that is not working? Or is it the screen?
Only touch dont work. 

Post web links to the vendors product page.  https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.6.2.373d4c4dzbHpLW&orderId=8140771111686037&productId=32985467436
The ILI9486 requires a slower TFT clock rate than the Ili9488. Try 10MHz.

Bodmer, 2021-11-28:
If you have 2 identical displays and one does not work in the exact same setup then I suspect one display must be faulty.  You will need to contact then vendor and ask for a replacement.

closed 2021-11-28 by Bodmer