Bodmer/TFT_eSPI #1459: White screen after (suspected) power fluctuation

cleishm, 2021-12-01:
Hi @Bodmer!

Firstly, thank you for the library and more for your active participation with the community of users.

I've built a small application powered by an ESP32 and a HX8357 that drives a relay/contactor**. While it works perfectly on the bench, about 20% of the time the display turns white when the contactor disengages (after a user-initiated "off" event occurs). The ESP32 is still functional, but the screen is not.

Given it works on the bench but not when attached to the contactor, I suspect a power fluctuation issue. I've tried powering the screen with the 5V coming directly from the power supply and also from the 3.3V line off the ESP32, with no apparent difference. I've tried dialing the SPI clock down, as low as 10M, and this also hasn't helped. Dialing it down to 5M seems to have helped, but it's probably too slow.

I'm at a bit of a loss at this point, and I'm unclear what I should attempt or even check for next. Without any other ideas, I'm at the point of seriously considering triggering a reset of the ESP32 whenever an "off" event occurs (I have confirmed that a reset does resolve the issue). I'm also going to look through your library to see if there is a way to reset just the screen.

I'd greatly appreciate any thoughts on what I could do, or where I could look for the issue next.

Thank you again!

** It actually drives a SainSmart Relay, which in turn powers the coil of a 240V contactor. The relay is optically isolated, with the relay coil powered by its own 5V supply.

Bodmer, 2021-12-01:
6. Add decoupling at the relay coil to suppress the inductive spikes from the relay coil.

cleishm, 2021-12-01:
And this is why you’re awesome. Thank you for helping me move forward!

closed 2021-12-01 by cleishm