Bodmer/TFT_eSPI #1460: Problem getting Rasp Pi Pico and Pico-ResTouch-LCD-3.5 working...

jas0995, 2021-12-01:
The Pico-ResTouch-LCD-3.5 is a display with a PICO socket on the back.  A very neat compact display package.

However, I have not been able to get it working for me.

I have copied Setup60_RP2040_ILI9341 and made changes specific to this display and saved it as Setup61_RP2040_ILI9488.

Then in User_Setup_Select.h the only commented setup is number 61.

I have tried several example programs that are included mainly focusing on the Read_User_Setup example.

I have tried this with both platformio and the Arduino IDE and in both cases setup() runs some Serial print code that i inserted and nothing else happens.  Regardless of which demo app I run. 

I assume that the IDEs are telling the code what processor I am using, because I don't see anywhere I need to specify it.

The key part of Setup61 I changed is:

```
// For the Pico use these #define lines
#define TFT_MISO  12
#define TFT_MOSI  11
#define TFT_SCLK  10
#define TFT_CS     9  // Chip select control pin
#define TFT_DC     8  // Data Command control pin
#define TFT_RST   15  // Reset pin (could connect to Arduino RESET pin)

 #define TFT_BL   13            // LED back-light control pin
 #define TFT_BACKLIGHT_ON LOW  // Level to turn ON back-light (HIGH or LOW)

#define TOUCH_CS  16     // Chip select pin (T_CS) of touch screen
```

Am I missing something?

Thanks
-John

jas0995, 2021-12-02:
While poking around with a o'scope, I noticed that the silk screen on the board did not match up with the pin numbers one the board docs....  Fixed that and now I have a new problem..

Now when I upload any program, I get four fast and four slow flashes of the led.

This seems to indicate the crashing of the mbed lib that the Arduino IDE uses.

Any ideas what may be causing this?

Thanks
-John

Bodmer, 2021-12-02:
The ones you have selected do not work. TFT_eSPI uses SPI port 0 (SPI0) in the hardware and this can only be allocated to certain pins. The example pins will work OK:
```
#define TFT_MISO  0
#define TFT_MOSI  3
#define TFT_SCLK  2
#define TFT_CS    4  // Chip select control pin
#define TFT_DC    5  // Data Command control pin
#define TFT_RST   6  // Reset pin (could connect to Arduino RESET pin)
//#define TFT_BL     // LED back-light

#define TOUCH_CS  7     // Chip select pin (T_CS) of touch screen
```

You need to check the Pico data sheet to find out which pins will work with SPI port 0. I will look too and see what the restrictions are. The official pinout for the Pico is here which indicates GPIO 16,17,18 and 19 are OK, I suspect there are other pin combinations that would be work but have not tested them.

Bodmer, 2021-12-02:
I have tested with these pins on a RPi Pico, and my ILI9341 SPI display works fine:
```
// For the Pico use these #define lines
#define TFT_MISO  16
#define TFT_MOSI  19
#define TFT_SCLK  18
#define TFT_CS    17  // Chip select control pin
#define TFT_DC    20  // Data Command control pin
#define TFT_RST   21  // Reset pin (could connect to Arduino RESET pin)

#define TOUCH_CS  22     // Chip select pin (T_CS) of touch screen
```
TFT_CS/DC/RST/BL and Touch_CS can be allocated to other pins, only GPIO 16, 1, 19 are important as they are valid SPI0 pins.

Kabron287, 2021-12-04:
I made exactly the same setup as **jas0995** did.

Execution cycled between **static inline uint spi_get_index** and **static inline spi_hw_t *spi_get_hw** subs in **spi.h** file.
**@jas0995**, what silkscreen  error did you find?

Bodmer, 2021-12-04:
@Kabron287 it appears you have a completely different problem to @jas0995

spi.h is not part of TFT_eSPI. It sounds like you have a custom SPI implementation so you will need to look more closely at that library.

Incidentally, TFT_eSPI does support 16 bit serial to parallel interfaces such as the MHS 4.0 Raspberry Pi TFT. One complication that may be relevant to you is that TFT commands are 8 bits on a normal SPI interface. So in TFT_eSPI, when RPI_DISPLAY_TYPE is defined the library will send 16 bit commands.

I am guessing the Raspberry Pi Pico bottom silk screen labels for GPIO have been misunderstood (e.g. GP14 and 15 positions) as they are deliberately shifted to miss the mounting holes, they are in the correct sequence. This is quite common on cramped boards and is not an error:

[image]

Bodmer, 2021-12-04:
I had a closer look at the waveshare LCD, I can see that the LCD is pre-wired to the SPI1 port so unfortunately TFT_eSPI cannot be used with that display at the moment. I suspect that a simple change could be done to support the SPI1 port so I will ad this to my TODO list.

Kabron287, 2021-12-04:
Thanks for attention **Bodmer**!
> only GPIO 16, 1, 19 are important as they are valid SPI0 pins.
But this board works with Waveshare examples and also with **Picomite MMBasic**.

> It sounds like you have a custom SPI implementation so you will need to look more closely at that library.
Not at all. I use exactly your library.

And what is important also, that should be stated out I guess: 
your library does not work with **https://github.com/earlephilhower/arduino-pico** package even if I manually add **ARDUINO_ARCH_RP2040** definition. 
Maybe due to aforementioned reasons.

Kabron287, 2021-12-04:
I confirm that ILI9341 works fine, but with rather strange output diagnostics.
[/code]
TFT_eSPI ver = 2.3.84
Processor    = Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO -119
MISO    = GPIO -116
SCK     = GPIO -118
TFT_CS   = GPIO -1
TFT_DC   = GPIO -1
TFT_RST  = GPIO -1

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 70.00
[/code]

Bodmer, 2021-12-04:
[Earle's RP2040 board package](https://github.com/earlephilhower/arduino-pico) seems to work fine for me when SPI0 pins are used.

I found a simple temporary fix to use SPI1 instead, this ONLY works with Phil's board package, simple add a new #define as follows:
```
#define spi0 spi1
#define TFT_MISO  12
#define TFT_MOSI  11
#define TFT_SCLK  10
#define TFT_CS     9  // Chip select control pin
#define TFT_DC     8  // Data Command control pin
#define TFT_RST   15  // Reset pin (could connect to Arduino RESET pin)

 #define TFT_BL   13            // LED back-light control pin
 #define TFT_BACKLIGHT_ON LOW  // Level to turn ON back-light (HIGH or LOW)

#define TOUCH_CS  16     // Chip select pin (T_CS) of touch screen
```
This has been tested on an ILI9341.

Bodmer, 2021-12-04:
I have updated the Read_User_Setup sketch to suit the RP2040, thanks for reporting that.

Bodmer, 2021-12-04:
The Pico-ResTouch-LCD-3.5 appears to use the same shift register design as the RPi boards.So it will need 16 bit commands instead of the normal 8 bit shifts. To invoke this add the following line to the beginning of your User_Setup file:

```
#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
```
Not that the shift register design on typically fails to work above 20MHz, so a lower frequency may need to be invoked.

jas0995, 2021-12-04:
Wow!  I go off to fix a 48,000 gallon water leak, and this thread takes off!! :)

@Bodmer, thanks for the work you do on this project!  
It looks like one of my biggest point of confusion may have been about the pinout of the Pico and the Display board.

When I first looked at the display and the pico, this is what I found...
[image]

and

[image]

At first I tried the GPxx numbers from the display, then when that did not work I switched to the pico pin numbers.  That is why there is some confusion on the i/o pin numbers I was using.  So with the Pico, I should be using the GPxx numbers.

I was planning on working over the code today to switch it from spi0 to spi1, but it looks like you beat me to it.

These are handy little boards available on Amazon, so it is good to hear that you have them on your TODO list to work on.

I will make changes to my Setup61_RP2040_ILI9488 file and let you know how it works.

Thanks for your work on this project!
-John

jas0995, 2021-12-04:
@Bodmer 
OK, It is working now!!

Here is my new setup file:  Setup61_RP2040_ILI9488.h
```
//                            USER DEFINED SETTINGS
//   Set driver type, fonts to be loaded, pins used and SPI control method etc
//
//   See the User_Setup_Select.h file if you wish to be able to define multiple
//   setups and then easily select which setup file is used by the compiler.
//
//   If this file is edited correctly then all the library example sketches should
//   run without the need to make any more changes for a particular hardware setup!
//   Note that some sketches are designed for a particular TFT pixel width/height

// ##################################################################################
//
// Section 1. Call up the right driver file and any options for it
//
// ##################################################################################

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
//#define TFT_PARALLEL_8_BIT

// Display type -  only define if RPi display
#define RPI_DISPLAY_TYPE // 20MHz maximum SPI

// Only define one driver, the other ones must be commented out
//#define ILI9341_DRIVER
//#define ST7735_DRIVER      // Define additional parameters below for this display
//#define ILI9163_DRIVER     // Define additional parameters below for this display
//#define S6D02A1_DRIVER
//#define RPI_ILI9486_DRIVER // 20MHz maximum SPI
//#define HX8357D_DRIVER
//#define ILI9481_DRIVER
//#define ILI9486_DRIVER
#define ILI9488_DRIVER     // WARNING: Do not connect ILI9488 display SDO to MISO if other devices share the SPI bus (TFT SDO does NOT tristate when CS is high)
//#define ST7789_DRIVER      // Full configuration option, define additional parameters below for this display
//#define ST7789_2_DRIVER    // Minimal configuration option, define additional parameters below for this display
//#define R61581_DRIVER
//#define RM68140_DRIVER
//#define ST7796_DRIVER
//#define SSD1963_480_DRIVER
//#define SSD1963_800_DRIVER
//#define SSD1963_800ALT_DRIVER
//#define ILI9225_DRIVER

// Some displays support SPI reads via the MISO pin, other displays have a single
// bi-directional SDA pin and the library will try to read this via the MOSI line.
// To use the SDA line for reading data from the TFT uncomment the following line:

// #define TFT_SDA_READ      // This option is for ESP32 ONLY, tested with ST7789 display only

// For ST7735, ST7789 and ILI9341 ONLY, define the colour order IF the blue and red are swapped on your display
// Try ONE option at a time to find the correct colour order for your display

//  #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//  #define TFT_RGB_ORDER TFT_BGR  // Colour order Blue-Green-Red

// For ST7789, ST7735 and ILI9163 ONLY, define the pixel width and height in portrait orientation
// #define TFT_WIDTH  80
// #define TFT_WIDTH  128
// #define TFT_WIDTH  240 // ST7789 240 x 240 and 240 x 320
// #define TFT_HEIGHT 160
// #define TFT_HEIGHT 128
// #define TFT_HEIGHT 240 // ST7789 240 x 240
// #define TFT_HEIGHT 320 // ST7789 240 x 320

// For ST7735 ONLY, define the type of display, originally this was based on the
// colour of the tab on the screen protector film but this is not always true, so try
[... 117 lines cut ...]
#define SPI_TOUCH_FREQUENCY  2500000

``` 
Here is the output of TFT_graphicstest_one_lib
```
17:20:45.394 -> TFT_eSPI library test!
17:20:45.792 -> Benchmark                Time (microseconds)
17:20:45.792 -> Screen fill              433939
17:20:46.225 -> Text                     45342
17:20:46.358 -> Lines                    352488
17:20:48.126 -> Horiz/Vert Lines         48322
17:20:48.260 -> Rectangles (outline)     25218
17:20:48.360 -> Rectangles (filled)      1051232
17:20:49.526 -> Circles (filled)         193651
17:20:49.826 -> Circles (outline)        173807
17:20:49.993 -> Triangles (outline)      77371
17:20:50.160 -> Triangles (filled)       382569
17:20:50.693 -> Rounded rects (outline)  69444
17:20:50.860 -> Rounded rects (filled)   1075347
17:20:52.026 -> Done!

```
If you would like me to run anymore tests, let me know
THANKS!!!
-John

Bodmer, 2021-12-04:
I have had a look at the official Arduino board package and there seems to be no easy way to persuade it to use SPI port 1 instead of port 0. There appear to be some hooks in place that refers to SPI1 but the code is incomplete.

So the at the moment SPI1 can only be used with [Earle Philhower's board package](https://github.com/earlephilhower/arduino-pico). I think that board package is supperior anyway as it supports dual core use easily, LittleFS and lots of board options. The [Arduino package](https://github.com/arduino/ArduinoCore-mbed) seems to be a WIP.

jas0995, 2021-12-05:
That works for me...   I use platformio anyway.  Unless I am testing something at least...

And Earle's board package can be used in platformio now.

-John

Kabron287, 2021-12-05:
Bingo!
It works on Waveshare !
Colors are inverted.

Bodmer, 2021-12-05:
I have updated my Arduino mbed package and now find the SPI1 mod works with that package too.

I will upload a fix later today.

Bodmer, 2021-12-05:
I have aded the option to define the port is the setup file:
```
// For the RP2040 processor define the SPI port channel used (default 0 if undefined)
//#define TFT_SPI_PORT 1 // Set to 0 if SPI0 pins are used, or 1 if spi1 pins used
```

closed 2021-12-05 by Bodmer

Kabron287, 2021-12-07:
Another strange thing, now concerning **ILI9341**.
Could be reproduced with **Read_User_Setup** with light modification (attached).
Compiled with Earle Philhower's board package as **Generic RP2040** gives SPI clock 24 MHz.
Compiled with Classic Arduino MBED package as **Raspberry PI Pico**  gives 64 MHz.
[Read_User_Setup.zip](https://github.com/Bodmer/TFT_eSPI/files/7668898/Read_User_Setup.zip)

zpm1066, 2022-01-04:
@Bodmer, The Pico Waveshare HATs work well on SPI1.
Thank you.

Bodmer, 2022-01-04:
@zpm1066 Thanks for letting me know.

@Kabron287 Unfortunately when the clock increases from125MHz the perpipheral clock gets attached to the USB clock at 48MHz, this means the SPI baud rate gets dropped from 62.5MHz to a maximum of 24MHz.  It is possible to boost the peripheral clock by attaching it to a higher frequency source but this messes up the baud rate for serial links apparently.

I will soon be releasing a PIO SPI implementation that will allow the SPI frequency to track the overclock frequency, then the processor performance gains from overclocking can be used to greater effect.

Bodmer, 2022-01-05:
THE SPI PIO imlementation is now in version 2.4.22 available direct from Github.

Bodmer, 2022-01-06:
I have found that it is possble to over-clock the RP2040 processor with Earle's board package and then increase the available SPI frequency by changing the clcok source to the UART and SPI. Add this code at the very start of setup() before the tft ot Serial ports are initialised:
```
 uint32_t freq = clock_get_hz(clk_sys);

 // clk_peri does not have a divider, so in and out frequencies must be the same
 clock_configure(clk_peri,
                    0,
                    CLOCKS_CLK_PERI_CTRL_AUXSRC_VALUE_CLK_SYS,
                    freq,
                    freq);
```
If you have overclocked to 250MHz this will mean that the maximum SPI frrequency is 125MHz but most displays will not work at this speed.

archearth, 2024-06-19:
I have gotten the display (Pico-ResTouch-LCD-3.5) to work with my pico W. However, I want to use the SD to load images/gif when a send a MQTT message to the pico. I was able to get to load images and gifs from LittleFS but when I try from the SD it renders in a glitchy slow way. I think the I dont have the right SPI and SD config for the SD and the display to work at the same time. Any ideas what pins to assign to the SD and how to configure it so the the TFT and SD both work at the same time? I am not very familiar with SPI and how all that works. Thanks!