Bodmer/TFT_eSPI #1465: Text not centered by 1 pixel vertically and horizontally...

exdiogene, 2021-12-03:
I have noticed that when TextDatum is CC_DATUM, there is always an offset of -1 pixel in both directions for any standard font size.

I am using a ESP32-WROVER-8 CPU on a WT32-SC01 V3.2 display board.

The screen is used in landscape mode.

You can test it with this code that use the standard 5X7 font in size 1 and 3 : 
```
    tft.init();
    tft.setRotation(1);
    tft.fillScreen(TFT_BLUE);
    tft.drawCircle(10,10,4,TFT_YELLOW);
    tft.drawLine(10,6,10,14,TFT_YELLOW);
    tft.drawLine(6,10,14,10,TFT_YELLOW);
    delay(5000);
    tft.setTextSize(1);
    tft.setTextColor(TFT_WHITE);
    tft.setTextDatum(CC_DATUM);
    tft.drawString("+",10,10);
    delay(5000);
    tft.drawCircle(40,40,14,TFT_YELLOW);
    tft.drawCircle(40,40,15,TFT_YELLOW);
    tft.drawCircle(40,40,16,TFT_YELLOW);
    tft.drawLine(39,25,39,55,TFT_YELLOW);
    tft.drawLine(40,25,40,55,TFT_YELLOW);
    tft.drawLine(41,25,41,55,TFT_YELLOW);
    tft.drawLine(25,39,55,39,TFT_YELLOW);
    tft.drawLine(25,40,55,40,TFT_YELLOW);
    tft.drawLine(25,41,55,41,TFT_YELLOW);
    delay(5000);
    tft.setTextSize(3);
    tft.drawString("+",40,40);
    delay(5000);

```

I always need to add "1" to the vertical and horizontal position to get it centered in buttons or elsewhere on the screen...

P.S. My previous issue is still not resolved, but it is still closed! Can you reopen it?

Bodmer, 2021-12-04:
Each Glyph has a bounding box, TFT_eSPI uses this to determine the centre. In the case of the 7x5 GLCD font the bounding box is 8x6. Since the Glyph could be anywhere in the box you can get offsets. For example a single full stop . will end up appearing to have an offset. 

Adding offsets in your code allows more precise content dependant positioning, so is the correct approach.

closed 2021-12-04 by Bodmer

exdiogene, 2021-12-04:
But the international standard is to leave 1 pixel free to the right and the bottom of the bounding box, so why do you not consider the logical approach to center the glyph that way?

The bottom free pixel is frequently used for the "underline" bar...

I just try to alleviate the burden to manually correct what was supposed to be correct with the use of CC_DATUM!

In my 50 years of coding I never saw a display using standard 5X7 font with the spacing pixel to the left and the top in all my projects... :(

So the logical standard would be to center on pixel x,y = 2,3 instead of 3,4 like you do for TextSize 1...

I was only trying to help here... ;)

Bodmer, 2021-12-04:
Help is always appreciated and there was no intent to discourage it.

At the pixel level, with simple bitmaps that have even pixel count boxes the glyph can't be centred on a single pixel unless anti-aliasing and hinting is used, that is beyond the scope of this library this means some tweaking might be needed.

There are all sorts of vagaries that you will find for historical Adafruit_GFX origin/compatibility reasons, as an example, even though the GLCD font is quoted as 5x7 by Adafruit many glyphs (e.g. comma and lower case desenders) occupy the 8th pixel row, so it is a 5x8 font.  In Adafruit_GFX the getTextBounds() function however returns w=6 and h = 8 for a GLCD character.

In your case x+1 is a good adjustment value, and y+1 might give better visual centreing with glyphs that do not use the 8th pixel line.

Other vagaries users have commented on include plotting a circle of radius 4 and finding it occupies 9x9 pixels, makes sense to have a central pixel for symetry reasons but a radius of 4.5 is not what a user might expect.