Bodmer/TFT_eSPI #1466: Touch response not working - ILI9488 driver based 3.5inch SPI TFT Display interfacing with Wemos Lolin32 (Esp 32 based ) MCU

Surendra-Gayke, 2021-12-04:
Hello Bodmer,
I am trying to interface 3.5" SPI TFT touchscreen to Wemos lolin32 (Esp wroom 32 based ) Microcontroller.
My Display program worked fine, But Touch program is not working.
Basically touchscreen is not responding to touch.

I am using following configurations is User_Setup.h file.

#define ILI9488_DRIVER 

#define TFT_MISO 19   // it is not connected to TFT T_DO
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS 5 // Chip select control pin
#define TFT_DC 2  // Data Command control pin
#define TFT_RST 16  // Reset pin (could connect to RST pin)
#define TOUCH_CS 15 

for reference I am attaching header file als
[User_Setup.txt](https://github.com/Bodmer/TFT_eSPI/files/7653718/User_Setup.txt)
o.

Bodmer, 2021-12-04:
Can you list all the wired connections you have used between the TFT display board and the ESP32.

Provide a web link to the display vendors page.

Surendra-Gayke, 2021-12-05:
Web link of display vendor page : https://www.probots.co.in/3-5-inch-tft-lcd-with-touch-screen-display-module-spi-interface.html

I am using Wemos lolin32 MCU. Attaching its pinouts.

Connections I have done
VCC & LED pin - 5V
GND - GND
SDI (MISO) & TOUCH_DO - 19 // Also tried to only connect to TOUCH_DO i.e. leaving SDI pin of TFT open
                                          
MOSI & TOUCH_DIN - 23  ( i.e. TFT_SDI & TOUCH_DIN made common & connected to GPIO 23 of wemos lolin32 )

SCLK & TOUCH_CLK -  18
TFT_CS- 5 // Chip select control pin
TFT_DC- 2 // Data Command control pin
TFT_RST- 16
TOUCH_CS- 15
TOUCH_IRQ is not connected

[image]

Bodmer, 2021-12-05:
The mistake is to connect SDI to MISO (if you meant SDO, then that is also a mistake.)

Only connect touch DO, to MISO and no other signals. Run the touch diagnostic sketch and post the results.

Confirm you have the serial only display and not the parallel option one as the vendors web page features two display types and conflicting descriptions. Post a picture of the back of the display board so I can see the touch control chip.

Bodmer, 2021-12-05:
I've tested my ILI9488 display in case recent changes have broken the touch interface. Mine works fine so we need to find out what is different in your configuration.

Surendra-Gayke, 2021-12-06:
Back of Display Picture :
[image]

Yes, Now working
By Connections you have suggested.
When I run Touch Calibration code,
I got these values.
uint16_t calData[5] = { 243, 3645, 318, 3461, 7 };
  tft.setTouch(calData);

But still I am unable to run Keypad code.
Can  you please test my code. Do I missing something in code? Attaching file
[Keypad.txt](https://github.com/Bodmer/TFT_eSPI/files/7659993/Keypad.txt)

Surendra-Gayke, 2021-12-06:
Also not able to run On-Off switch code

Bodmer, 2021-12-06:
The Keypad_480x320 and ON/OFF sketches should run a calibration then display the keypad or switch. Do you get the calibration screen and touch the 4 corners as requested?

Surendra-Gayke, 2021-12-10:
Sorry for delay in response.
Yes I got the calibration file.
Now On/Off sketch is working.

But still keypad sketch is not working.

I got calibration file as
uint16_t calData[5] = { 243, 3645, 318, 3461, 7 };
  tft.setTouch(calData);

Bodmer, 2021-12-11:
Great, this means touch is working so it is just down to the calibration for each sketch.

You probably need to repeat the calibration in the keypad sketch by setting the line in the sketch as follows:
```
#define REPEAT_CAL true
```
This will force a new calibration file to be saved in SPIFFS.

Calibration has to be done in the screen rotation you will use. Bear this in mind if you edit the sketch.

The calibration array calData[] is for processors that do not have SPIFFS and are not needed for the ON/OFF and Keypad examples.

closed 2021-12-11 by Bodmer

alvarotorijano, 2022-04-14:
hi, would be possible to use the touch support using the 8bit paralel interface board versions?

Bodmer, 2022-04-14:
Yes, but you have to use a third party library. There are libraries for the many different touch interface types found on parallel displays (I2C, SPI, analogue). For resistance based analogue I/O interface screens (typically UNO style) on an ESP32 to 8 bit parallel screen check the ReadMe for a link.

alvarotorijano, 2022-04-17:
Thank you very much for such a good job and for answering so fast. Found the link you mention and tested with ILI9486 and ILI9481 Arduino style 8bit interface modules with both of them working. For any reason, I did not see that point on the main page which is clearly visible. Sorry.
Thank you again!

MorisB, 2023-01-05:
First of all, thank you for your investment and sharing.
I have a problem with touch, on the ILI9488 display that includes a touch chip.
I am connecting an ESP32 board(Lolin32), for display with your library,
When I run graphics codes, I can see graphics, and works perfect.
But, with the touch code, I can't see any touch options, I get a white screen.
Where am I wrong?

ash20242024, 2024-03-27:
May I know the User_Setup_Select chosen for this display with ESP32 ?