Bodmer/TFT_eSPI #1470: No Issue - An idea how to speed up the drawing of lines

GeorgHelmut, 2021-12-04:
Hi, I just joining the community, hence be patient....

Since some days I'm diving into the world of TFT's, hardware and software. Looking into the standard method of drawing free lines by using Bresenham Algorithm I got an idea how to simplify and speed up this time consuming routine. Finally - why not to share the solution with the community for discussion.   

[Line_en.pdf](https://github.com/Bodmer/TFT_eSPI/files/7654360/Line_en.pdf)

GeorgHelmut, 2021-12-04:
By the way: 
While developing this Line Code I came across that my ILI 9486 chip has a glitch at column 255. When drawing a H-Line at this position two ascending pixels are highlighted. I got close to crazy as I thought it's caused by my coding. But multiple tests have proven that the chip has a glitch. There is no manufacturer code readable from this chip. May be, a copied design? 
Did anyone else come across such a issue?

Bodmer, 2021-12-04:
Hi, thanks for the suggestion. I will move to the "Discussions" area as it is more of a topic of interest than an issue.

The algorithm works well and runs fast. I usually run the "graphicstest" sketch to check for performance deltas on the algorithms.

There are a few anomlies in the line at the ends, I think this may be because the end pixels should be excluded in calculations.

Existing algorithm plot from graphicstest:

[image]

The suggested code output:

[image]

These look very similar until the bottom and right edges are inspected closely where there is a large final adjustment to the line in the new algorithm:

[image]

Bodmer, 2021-12-04:
Tricky to see without zooming in:
[image]

Bodmer, 2021-12-04:
The 45 deg angle also has this last correction. The line should consist of single pixels:

[image]

I think a simple tweak to the algorithm is all that is needed.