Bodmer/TFT_eSPI #1472: Any way to use ILI9481 IPS SPI display using TFT_eSPI

jtokt, 2021-12-05:
Looking into User_Setup_select.h, User_Setup.h and setupfiles in UserSetups,  I tried to use a combination of ILI9481 driver and Setup21_ILI9488.h which switched from #define ILI9488_DRIVER to #define ILI9481_DRIVER.  But it does not work.

Is there any way to use ILI9481 IPS SPI display using TFT_eSPI ?

Bodmer, 2021-12-05:
Yes, the ILI9481 should work fine. It is quite a slow chip so try 10MHz to start with and increase slowly to find the limit if that works.

Bodmer, 2021-12-05:
Setup 51 is for the LilyPi board and has been tested by a user at 26.66MHz.

jtokt, 2021-12-05:
I appreciate for your prompt reply．I tried 10MHz and successfully displayed.  For the community's beniefit, I increased the clock such as 12.5MHz, 15.0MHz and 17.5MHz.   Up to 15.MHz, display were fine.  17.5MHz did not work.  

Thanks again Bodmer.

Bodmer, 2021-12-05:
Thanks, that is useful to know.

closed 2021-12-05 by Bodmer