Bodmer/TFT_eSPI #1475: drawString, screen artefacts while reading touch

sfxsfl, 2021-12-07:
### Discussed in https://github.com/Bodmer/TFT_eSPI/discussions/1474

<div type='discussions-op-text'>

<sup>Originally posted by **sfxsfl** December  6, 2021</sup>
Hi, 

first of all. Thank you for the great piece of work and the effort you spend developing it. 
I'm using the ESP8266 with an ili9341 with Touch in SPI Overlap mode. 

In my simple sketch I'm using drawString within the setup and getTouch in loop. As a result
I'm getting artifacts on the screen at the last character. 

`
#include <TFT_eSPI.h> 
TFT_eSPI tft = TFT_eSPI();  // Invoke library

void setup(void) {
tft.init();
  tft.setRotation(1);
  tft.fillScreen(TFT_DARKGREY);
  tft.setTextColor(TFT_WHITE,TFT_BLACK);  tft.setTextSize(1);
  // We can now plot text on screen using the "drawString"
  tft.drawString("Welcome the ESP8266", 100, 110, 2); 
}

void loop() {
  uint16_t x,y; 
#if 1
  if (tft.getTouch(&x, &y)) {   
    delay(100); 
  }
#endif
delay(200);  
}
`
UserSetup.h
`
#define ILI9341_DRIVER
#define TFT_SPI_       OVERLAP
#define TFT_CS         PIN_D3
#define TFT_DC        PIN_D1
#define TOUCH_CS   PIN_D4  // Chip select pin (T_CS) of touch screen
#define TFT_RST      -1    // Set TFT_RST to -1 if the display RESET is connected to NodeMCU RST or 3.3V
 
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
//#define LOAD_FONT8N // Font 8. Alternative to Font 8 above, slightly narrower, so 3 digits fit a 160 pixel TFT
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// Comment out the #define below to stop the SPIFFS filing system and smooth font code being loaded
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

#define SPI_FREQUENCY  27000000
// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000
// #define SPI_TOUCH_FREQUENCY  1000000
`

Thanks in advance
BR Stefan

 </div>

sfxsfl, 2021-12-07:
[image]

sfxsfl, 2021-12-07:
Hi,
I've verified the problem with TFT_ESPI_VERSION "2.3.87" still the same. However, in case you will give a try, you may use 
tft.getTouchRawZ() and it also creates artifacts on the screen on the last character. It boils down to spi.transfer function supplied by <SPI.h>. 

Thanks in advance
Stefan

Bodmer, 2021-12-08:
The problem is that in Overlap mode only one SPI device is permitted on the bus. I can see that I forgot to mention this in the setup file.

The only way around this is to avoid overlap mode is a tocuh screen is used.

I will update the notes in the setup file so this is clear.

closed 2021-12-08 by Bodmer

Bodmer, 2021-12-08:
The DC line is the data/command line to the TFT so altering the state of this line will change the fault signature since garbage will then end up in the TFT command registers rather than the display data RAM. This is likely to cause more unwanted effects. The CS line (D3) to the TFT will unfortunately get toggled low when you write/read touch request commands, this is under control of the ESP8266 to avoid conflicts with the FLASH read/write access so do not try to take control of the TFT CS line.

On option is to add a multiplexer that selects whether the D3 GPIO pin selects the TFT or touch controller.