Bodmer/TFT_eSPI #1477: An invalid Argument for tft.setfont() is bricking STM32F103C8

imalipusram, 2021-12-08:
When supplying tft.setfont() with an invalid argument, I did tft.setfont(10), the STM32F103C8 (aka BluePill) becomes bricked and requires erasing the flash with STM32CubeProgrammer from the ST website, to be made accessible again. 

I'm using an ILI9341 hooked up to SPI2 and and uploading code from the Arduino IDE with a ST-Link V2 clone. 

The issue very likely might be circumvented by checking the argument of tft.setfont() before passing it to the function.

Bodmer, 2021-12-08:
This should not happen as the parameter validity is trapped during font rendering.

I have tried a sketch with invalid argument and the STM32F103C8 (aka BluePill) continues to run, albeit not printing to the screen as the font number is invalid.

This is my test sketch, please try it.

```
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();

void setup(void) {
  Serial.begin(115200);
  Serial.println("\n\nStarting...");

  tft.init();
  tft.fillScreen(TFT_BLACK);
  tft.setTextDatum(MC_DATUM);
  tft.setTextSize(1);
  //tft.setTextColor(TFT_WHITE,TFT_BLUE);
  tft.setTextColor(TFT_WHITE);

  tft.setTextFont(10); // <<<-----<<<< Invalid argument test

  for(char i=32;i<128;i++) {
    tft.fillScreen(TFT_BLACK);
    String str =  String((char)i);
    tft.drawString(str,160,240);
    Serial.println(str);
    delay(100);
  }

  while(1) delay(10);
}

void loop() {
}
```

Bear in mind that due to the limited FLASH resources of the Blue Pill you will have to comment out some fonts. In my test I had two fonts loaded:
```
#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
```
The limited RAM also means that only tiny Sprites can be used and there may be insufficient resources for Smooth fonts.

Since I cannot reproduce the problem, can you provide a sketch that shows the problem and provide a copy of the setup file.

Bodmer, 2021-12-11:
I have not managed to reporoduce this and without a n example sketch that causes problem I will not be able to progress. I will close but feel free to ask more questions if you need help.

closed 2021-12-11 by Bodmer