Bodmer/TFT_eSPI #1481: Landscape mode for ST7789V 320x240 screens

6b32, 2021-12-09:
Hi, is it possible to make landscape mode for these drivers work? I have 320x240 screens with this driver and it will just never work on full 320x240 in landscape or portrait. There is always grey area and active resolution is 240x240 no matter what. 

With ILI9431 it works no problem, 320x240 portrait  or landscape. 

Thanko you.

Bodmer, 2021-12-09:
It sound like the pixel dimentsions in the setup file are not set correctly. They must be set in protrait mode [as noted here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L77), e.g.:
#define TFT_WIDTH  240
#define TFT_HEIGHT 320

6b32, 2021-12-09:
Been battling that for 3 days, tried all combinations.. Today I received other TFT and it works. Oh.
So for anyone with same issue, try other TFT screen. For me:

2" IPS 320x240 from DFrobot did **not work.**
2.4" IPS 320x240 from DISPLAY ELEKTRONIK with red tab on screen works.

Sorry to bother you.

Bodmer, 2021-12-10:
No problem glad you have a working setup.

closed 2021-12-10 by Bodmer