Bodmer/TFT_eSPI #1484: Display problem on a TTGO T-display

kalon33, 2021-12-12:
Using a TTGO T-display, I got only part of the display used, with the remaining of the display with glitches. T-display is properly chosen in `User_Setup_Select.h`.

I tried rotating without success. It also seems that the colors are inverted, as I should got that result:

[image]
 
But got this one:
[image]

For more info about my board, I took a picture from the other side:
[image]

Could you help me fix that problem? Thanks!

Bodmer, 2021-12-12:
I have an identical looking board and it seems to work fine with setup 25 and the library examples.

To test for colour run the Test and diagnostics example Colour_Test. This is what you should see:

[image]

To check your setup file is correctly selected run the Read_User_Setup example and look at the serial monitor output.  Here's a copy of what I see when I do this (with latest Github version of library):
```
TFT_eSPI ver = 2.3.88
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 7789
Display width  = 135
Display height = 240

R0 x offset = 52
R0 y offset = 40
MOSI    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 5
TFT_DC   = GPIO 16
TFT_RST  = GPIO 23

TFT_BL           = GPIO 4
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
```
If this looks good then I will need a complete example sketch that demonstrates the problem to see where the problem is.

kalon33, 2021-12-13:
[code]
TFT_eSPI ver = 2.3.70
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = SPI
Display driver = 9341
Display width  = 240
Display height = 320

MOSI    = GPIO 19
SCK     = GPIO 18
TFT_CS   = GPIO 5
TFT_DC   = GPIO 16
TFT_RST  = GPIO 23

TFT_BL           = GPIO 4
TFT_BACKLIGHT_ON = HIGH

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.00
[/code]

Could it be possible that I don't have the appriopriate driver?

kalon33, 2021-12-13:
For the color test, I got that:
[image]
[image]

Bodmer, 2021-12-13:
The following lines indicate the setup file is not correct:
```
Display driver = 9341
Display width = 240
Display height = 320
```
This should read the same as mine, viz:
```
Display driver = 7789
Display width  = 135
Display height = 240

R0 x offset = 52
R0 y offset = 40
```
Probably [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L20-L22) has not been commented out?  Typically this leads to warning message during compile time but you will need to check that only ONE setup file is being called up.

kalon33, 2021-12-13:
@Bodmer : That was the problem... Sorry... Thanks a lot for your help!

Bodmer, 2021-12-13:
No problem. Have made the same mistake myself!

closed 2021-12-13 by Bodmer