Bodmer/TFT_eSPI #1486: ILI9341 320x240 Screen suddenly turned inverted colors after few mins of running

LiveRock, 2021-12-13:
I am using an ESP32 with ILI9341 320x240 Touch Screen, 2 sensors and ESP-NOW, and also communication via Serial.

ESP-NOW is on core 0 and the rest on core 1.
The firmware runs ok for a few mins and then, the LCD turned inverted colors;  i.e. black becomes white, green becomes purple, etc.

I had spent the last 10 days trying all sorts of ways from Tasks Scheduler, Simple Multi-Tasking, etc. but still can't fund the source of the issue. This issue is driving me nuts...

HELP!

LiveRock, 2021-12-13:
I am not using Task Scheduler, or vTaskPintoCore. I am just Simple Multi-Tasking
And I have also disabled the WatchDog for both cores.

Bodmer, 2021-12-13:
I am not familiar with ESP NOW although I do know what it does.

It will not be easy to debug without a detailed knowledge of how you are using the two taks and how they might be interacting.

Typically these messaging systems use a call back and the data will need to be stored and passed to the other processor. Only one processor can handle the TFT so the tasks need to be well partitioned.

Bear in ind that TFT_eSPI is intended for a single tasking envionment, the ESP32 is actually running an RTOS in the background so maybe these two tasks just won't cooperate to properly manage the different acitivites.

So it will need to be debugged by test and inspection to identify the root cause.

Out of interest I found this link so might have a play with ESP NOW at some point:
https://randomnerdtutorials.com/esp-now-esp32-arduino-ide/

I will move this to a discussion as there is no identifyable issue I can see with TFT_eSPI.