Bodmer/TFT_eSPI #1493: .VLW fonts cannot be used !?

deruiter, 2021-12-16:
Used:
Arduino Nano 33 IoT (SAMD21) 
IDE: PlatformIO

If I try to use a freshly made **.vlw font** _(OR one I made and used months ago)_ the compiler goes crazy
The ones I used with another project was for a ESP32 board, that did work!?

compiler output:

`\xff \xfa\xff
                                                                                                                                                                                                                                                                                                                                                                ^
include/Final-Frontier20.vlw:47:353: warning: null character(s) ignored
  \x93\xf4  \xe8\xff9 pO   \xa0\xff\xd7C  \xe6\xff\xff\xff\xff\xff\xd6 \x96\xe0\xfb\xff\xff]     >D
  !\xc8\xff\xff\xd5  \xbb\xff\x92fH   \xf6\xff     \xff\xff    \xde\xff\xff\xff\xff{ =\xff\xff\xff\xff\xf2   \xff\xff %    \xff\xff      \xff\xff      \xff\xff      \xff\xff      \xff\xff     "\xff\xff"    r\xca\xf3\xff\xff\xff\xff(
                            \xc8\xff\xff\xff\xff\xff\xff\xff \x88\xff\xdfO  \xff\xff \xdf\xff>    \xff\xff \xfa\xff
                                                                                                                                                                                                                                                                                                                                                                 ^
include/Final-Frontier20.vlw:47:363: error: stray '\377' in program
  \x93\xf4  \xe8\xff9 pO   \xa0\xff\xd7C  \xe6\xff\xff\xff\xff\xff\xd6 \x96\xe0\xfb\xff\xff]     >D
  !\xc8\xff\xff\xd5  \xbb\xff\x92fH   \xf6\xff     \xff\xff    \xde\xff\xff\xff\xff{ =\xff\xff\xff\xff\xf2   \xff\xff %    \xff\xff      \xff\xff      \xff\xff      \xff\xff      \xff\xff     "\xff\xff"    r\xca\xf3\xff\xff\xff\xff(
                            \xc8\xff\xff\xff\xff\xff\xff\xff \x88\xff\xdfO  \xff\xff \xdf\xff>    \xff\xff \xfa\xff
                                                                                                                                                                                                                                                                                                                                                                           ^
*** [.pio/build/nano_33_iot/src/main.cpp.o] Build interrupted.
Error: Aborted by user
`

SKETCH:

`
#include <Arduino.h>
// Wifi
#include <TFT_eSPI.h> // Include the graphics library (this includes the sprite functions)
//
#include <SPI.h>
#include <Wire.h>

#include "NotoSansBold15.h"
#include "NotoSansBold36.h"
#include "Final-Frontier20.vlw"

// The font names are arrays references, thus must NOT be in quotes ""
#define AA_FONT_SMALL NotoSansBold15
#define AA_FONT_LARGE "Final-Frontier20.vlw"

#include <SPI.h>
#include <TFT_eSPI.h> // Hardware-specific library

TFT_eSPI tft = TFT_eSPI();

void setup(void)
{
  tft.begin();
  tft.setRotation(2);
}
void loop()
{

  tft.fillScreen(TFT_BLACK);
  tft.setTextColor(TFT_WHITE, TFT_BLACK); // Set the font colour AND the background colour
                                          // so the anti-aliasing works

  tft.setCursor(0, 0); // Set cursor at top left of screen
  // >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
  // Small font
  // >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
  tft.loadFont(AA_FONT_SMALL); // Must load the font first
  tft.println("Small font");   // println moves cursor down for a new line
  tft.unloadFont();
  delay(1000);

  // >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
  // Small font
  // >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
  tft.loadFont(AA_FONT_LARGE); // Must load the font first
  tft.println("LARGE font");   // println moves cursor down for a new line
  delay(1000);
}
`

closed 2021-12-16 by deruiter

Bodmer, 2021-12-18:
The font must be included.

deruiter, 2021-12-18:
Thank you for your reply but I closed the issue?!

 because after hours of searching I found that I had to convert the .vlw fonts!

This was NOT the case with my previous use of the TFT_eSPI library and .vlw fonts in another 8266 project so I didn't think about that.

Maybe it is useful to include in the readme that .vlw fonts has to be converted.

Bodmer, 2021-12-18:
The examples should have the vlw font included in the sketch. Did you find a sketch with missing vlw font files?

deruiter, 2021-12-18:
In the samples I only see (two)  *.h font files so those are the _converted_ .vlw files.?

As I said, I did not realize fast enough that I can not use .vlw files but need to convert them to .h files by using the web site you mentioned in the .h font file.
  
When I did not work I looked back into an few month old project and there I could use 'raw' .vlw files just by calling them??  That was what kept me trying and failing.

So I think you have upgraded the TFT_eSPI library so one can only use converted .h files?

deruiter, 2021-12-18:
I felt stupid for asking so I closed the issue as soon as possible after discovering the solution in one of the converted .h font files...

Bodmer, 2021-12-18:
The vlw files only need to be converted to .h files if the fonts are to be stored in FLASH memory directly as an array. This is a new(ish) feature, in old versions of the library the only option was to keep a raw .vlw file within a filing system e.g. SPIFFS, LittleFS or SD card. The new feature was added because some processors (e.g. STM32) do not have a FLASH based filing system available.

I will look to make this cleared in the readMe.

deruiter, 2021-12-18:
ooooh!  ok... I didn't realize that!   But it's good to mention that in the readme, thank you!

Bodmer, 2021-12-18:
ReadMe has been updated to be clearer!

Thanks for reporting this.