Bodmer/TFT_eSPI #1494: Use this setup in parallel with touch fonction?

Kryaak, 2021-12-16:
Hello,

I'm novice in this world and I wanted to make some Cyrptocurrency display on ​Mcufriend shield screen (ili9341) (red one) for uno, with espduino32 (https://www.cdiscount.com/pdt2/6/3/4/1/700x700/vod0721047383634/rw/carte-de-developpement-espduino-32-esp32-wifi-bl.jpg)

TFT_eSPI ver = 2.3.70
Processor    = ESP32
Frequency    = 240MHz
Transactions = Yes
Interface    = Parallel
Display driver = 9341
Display width  = 240
Display height = 320
 
TFT_CS   = GPIO 33
TFT_DC   = GPIO 15
TFT_RST  = GPIO 32
TFT_WR   = GPIO 4
TFT_RD   = GPIO 2

TFT_D0   = GPIO 12
TFT_D1   = GPIO 13
TFT_D2   = GPIO 26
00:TFT_D3   = GPIO 25
TFT_D4   = GPIO 17
TFT_D5   = GPIO 16
TFT_D6   = GPIO 27
TFT_D7   = GPIO 14

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define ILI9341_DRIVER       // Generic driver for common displays
#define TFT_PARALLEL_8_BIT

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   15  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    4  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    2  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   17
#define TFT_D5   16
#define TFT_D6   27
#define TFT_D7   14

Can you say me if it's possible to use this setup in parallel with touch fonction?

I have this compiling error: 'class TFT_eSPI' has no member named 'getTouch' 

Thanks for your support. 
Happy christmas.

Bodmer, 2021-12-17:
The library does not support touch when a parallel display is used. This is [explained here](https://github.com/Bodmer/TFT_eSPI#8-bit-parallel-supportset) and there is a link at the end of that section to a touch library that can be used.

closed 2021-12-17 by Bodmer

Kryaak, 2021-12-17:
Thanks you for supporting noobs :)
Have nice christmas holiday.