Bodmer/TFT_eSPI #1497: Passing library as a pointer.

builderdev212, 2021-12-18:
I'm having an issue with declaring the library as a pointer, to use it in a class as shown below.
```c++
class test {
    public:
        test(TFT_eSPI* _tft) {
            tft = _tft
        }
    private:
        TFT_eSPI* tft;
}
```
However, when I initialize the library as a pointer, it returns an error.
Code:
```c++
TFT_eSPI *tft = TFT_eSPI();
```
Error:
```
Arduino: 1.8.14 Hourly Build 2021/04/08 03:25 (Linux), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 1, None"

menu:5:26: error: cannot convert 'TFT_eSPI' to 'TFT_eSPI*' in initialization
 TFT_eSPI *tft = TFT_eSPI();

exit status 1
cannot convert 'TFT_eSPI' to 'TFT_eSPI*' in initialization
```
I'm not completely sure if there is a syntax that this could work with, hence why I've made this issue. Any suggestions/advice about how I might create a pointer for it, or the fact that it may not be possible would be much appreciated! 

Once again, thank you in advance. 
Merry Christmas!

Pablo2048, 2021-12-18:
try
```cpp
TFT_eSPI *tft = new TFT_eSPI();
```

builderdev212, 2021-12-18:
Aha, that works! Thank you for the help. :D

closed 2021-12-18 by builderdev212