Bodmer/TFT_eSPI #1499: TFT_eSPI.cpp:3289:21: error: 'SPI_X' was not declared in this scope

alf45tar, 2021-12-19:
Latest commit create an issue during compile

https://github.com/alf45tar/PedalinoMini/issues/299#issue-1084044581

Does it depends by library or my code?

Thanks
alf45tar

melkati, 2021-12-19:
I have this issue also in [my project](https://github.com/melkati/CO2-Gadget/tree/bug-Bodmer/TFT_eSPI%402.3.70). 
I have reverted to [my fork](https://github.com/melkati/TFT_eSPI) with 2.3.70 and it's working fine.

Using a TTGO T-Display
```

	-DST7789_DRIVER=1
	-DENABLE_TFT=1
	-DTFT_WIDTH=135
	-DTFT_HEIGHT=240
	-DCGRAM_OFFSET=1
	-DTFT_MOSI=19
	-DTFT_SCLK=18
	-DTFT_CS=5
	-DTFT_DC=16
	-DTFT_RST=23
	-DTFT_BL=4
	-DTFT_BACKLIGHT_ON=1
	-DLOAD_GLCD=1
	-DLOAD_FONT2=1
	-DLOAD_FONT4=1
	-DLOAD_FONT6=1
	-DLOAD_FONT7=1
	-DLOAD_FONT8=1
	-DLOAD_GFXFF=1
	-DSMOOTH_FONT=1
	-DSPI_FREQUENCY=40000000
	-DSPI_READ_FREQUENCY=6000000

```

Bodmer, 2021-12-19:
Thanks for reporting this. Should be fixed now.

commit e66d0f40 (2021-12-19): Fix #1499
closed 2021-12-19 by Bodmer