Bodmer/TFT_eSPI #1501: 2.4.0 breaks PlatformIO build for some platforms/frameworks

rafl, 2021-12-19:
I have an existing PlatformIO application that builds against and works correctly with TFT_eSPI 2.3.89. This is its `platformio.ini`:

```
[env:wemos_d1_mini32_knockoff]
platform = espressif32
board = wemos_d1_mini32
framework = arduino
lib_deps = bodmer/TFT_eSPI@~2.3.89
build_flags =
  -DUSER_SETUP_LOADED=1
  -DILI9488_DRIVER=1
  -DTFT_CS=15
  -DTFT_MOSI=13
  -DTFT_SCLK=14
  -DTFT_MISO=12
  -DTFT_DC=21
  -DTFT_RST=17
  -DLOAD_GLCD
  -DLOAD_FONT2
  -DLOAD_FONT4
  -DLOAD_FONT6
  -DLOAD_FONT7
  -DLOAD_FONT8
  -DLOAD_GFXFF
  -DSMOOTH_FONT
  -DUSE_HSPI_PORT
  -DSPI_FREQUENCY=40000000
  -DSPI_READ_FREQUENCY=6000000
```

When updating to the latest TFT_eSPI version (by changing `~2.3.89` to `~2.4.0` in `lib_deps`), the application will no longer build:

```
.pio/libdeps/wemos_d1_mini32_knockoff/TFT_eSPI/TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::drawPixel(int32_t, int32_t, uint32_t)':
.pio/libdeps/wemos_d1_mini32_knockoff/TFT_eSPI/TFT_eSPI.cpp:3289:21: error: 'SPI_X' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                     ^
.pio/libdeps/wemos_d1_mini32_knockoff/TFT_eSPI/TFT_eSPI.cpp:3289:26: error: 'spi_get_hw' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                          ^
.pio/libdeps/wemos_d1_mini32_knockoff/TFT_eSPI/TFT_eSPI.cpp:3289:34: error: 'SPI_SSPSR_BSY_BITS' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                                  ^
.pio/libdeps/wemos_d1_mini32_knockoff/TFT_eSPI/TFT_eSPI.cpp:3294:20: error: 'SPI_X' was not declared in this scope
     spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
                    ^
.pio/libdeps/wemos_d1_mini32_knockoff/TFT_eSPI/TFT_eSPI.cpp:3294:32: error: 'spi_cpol_t' was not declared in this scope
     spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
                                ^
[ ... further similar errors omitted ... ]
```

It looks like some RP2040-specific code might be included with that particular configuration on the newer version.

pingalit, 2021-12-19:
Adding to this issue:

Previously, I was able to compile using version 2.3.70. I'm currently compiling using v2.4.0, and now I'm getting compile errors.
Here's my current platform.ini:
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	bodmer/TFT_eSPI@^2.3.70
	hideakitai/ESP32DMASPI@^0.1.2
	joaolopesf/SerialDebug@^0.9.82
```

Here's the dependency graph:
```
|-- <TFT_eSPI> 2.4.0
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <ESP32DMASPI> 0.1.2
|   |-- <SPI> 1.0
|-- <SerialDebug> 0.9.82
|-- <SPI> 1.0
```

This is the output:
```
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp: In member function 'virtual void TFT_eSPI::drawPixel(int32_t, int32_t, uint32_t)':
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3289:21: error: 'SPI_X' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                     ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3289:26: error: 'spi_get_hw' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                          ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3289:34: error: 'SPI_SSPSR_BSY_BITS' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                                  ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3294:20: error: 'SPI_X' was not declared in this scope
     spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
                    ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3294:32: error: 'spi_cpol_t' was not declared in this scope
     spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
                                ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3294:47: error: 'spi_cpha_t' was not declared in this scope
     spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
                                               ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3294:61: error: 'SPI_MSB_FIRST' was not declared in this scope
     spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
                                                             ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3294:74: error: 'spi_set_format' was not declared in this scope
     spi_set_format(SPI_X,  8, (spi_cpol_t)0, (spi_cpha_t)0, SPI_MSB_FIRST);
                                                                          ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3296:19: error: 'spi_get_hw' was not declared in this scope
   spi_get_hw(SPI_X)->dr = (uint32_t)TFT_CASET;
                   ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3298:34: error: 'SPI_SSPSR_BSY_BITS' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS){};
                                  ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3305:34: error: 'SPI_SSPSR_BSY_BITS' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                                  ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3309:34: error: 'SPI_SSPSR_BSY_BITS' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                                  ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3316:34: error: 'SPI_SSPSR_BSY_BITS' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                                  ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3328:36: error: 'SPI_SSPSR_BSY_BITS' was not declared in this scope
     while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                                    ^
.pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:3352:34: error: 'SPI_SSPSR_BSY_BITS' was not declared in this scope
   while (spi_get_hw(SPI_X)->sr & SPI_SSPSR_BSY_BITS) {};
                                  ^
```

For now, I will revert to a previous version that was stable for my application.

Bodmer, 2021-12-20:
This should be fixed in 2.4.01

closed 2021-12-20 by Bodmer